freerouting/freerouting #127: Proofreading and translations

andrasfuchs, 2022-07-05:
I created a [public project page for freerouting on Crowdin](https://crowdin.com/project/freerouting).

I invite everyone to join who can spare a few hours to make and proofread our translations.

Currently we have English, German and Simplified Chinese support, and I'm looking for people who can help with the following additional languages:
* Hindi
* Spanish
* French
* Arabic
* Bengali
* Russian
* Portuguese

Every forthcoming major version of freerouting will have updated translations for the languages that hit the 95% completion level.

andrasfuchs, 2022-08-01:
Thank you @JonathSpirit for helping the French translation effort!

andrasfuchs, 2022-09-22:
Translations were completed by multiple AI-based translator services for Hindi, Spanish, French, Arabic, Bengali, Russian, Portuguese, Japanese and Korean. Ideally these translations would be proofread by native speakers, but the more likely way will be that I'll include them in the next release, and wait for feedback from the community.

If anyone feels like looking into this, please check out https://freerouting.crowdin.com/

andrasfuchs, 2022-10-13:
The following languages are now supported: Spanish, French, Arabic, Russian, Portuguese and Japanese.

The following are translated, but they seems to be not working as expected: Hindi, Bengali and Korean.

Any help with these were greatly appreciated!

andrasfuchs, 2022-10-20:
Hindi, Bengali and Korean have problem with the font we are using. It needs to be investigated what font should we replace it with.

NirmalChauhan, 2023-06-08:
> The following languages are now supported: Spanish, French, Arabic, Russian, Portuguese and Japanese.
> 
> The following are translated, but they seems to be not working as expected: Hindi, Bengali and Korean.
> 
> Any help with these were greatly appreciated!

I would love to help with Hindi Translation. I have just started learning JAVA and I would love to help. 
I was just browsing GitHub and I just landed on this repository. So I might face some problems as I am new, but I will try.

andrasfuchs, 2023-06-09:
Thank you @NirmalChauhan,  I appreciate it!

What is the exact error message when you try to access Crowdin? Does the website work at all?
(I could test it from an Indian IP address, and it worked fine for me.)

NirmalChauhan, 2023-06-09:
> Thank you @NirmalChauhan, I appreciate it!
> 
> What is the exact error message when you try to access Crowdin? Does the website work at all? (I could test it from an Indian IP address, and it worked fine for me.)

So, Sorry Actually I created a developer account in Crowdin, Now I can access after making the enterprise account. I didn't create it earlier as I thought it would charge me.

rockingdemon, 2023-09-29:
Hello @andrasfuchs and great thanks for your work.
I'm trying to help this project with Russian translation. I've tried recent version and my eyes are bleeding from machine translation ;-)
I think I'll perform several phases of this task.
At first I'll try my best to translate resource bundles "as is". I mean, I'm a native Russian, but not such a PCB pro. Anyway, some cleanup will help and I'm already seeing good results.
At second phase I think different approach to message generation should be applied. I've noticed that sometimes there are direct string concatenation and this works really bad for translation. It seems message templating would fit better.

andrasfuchs, 2023-10-03:
Hello @rockingdemon!

Thanks for the feedback! Yes, I can imagine, machine translation can be terribly wrong with some languages :)

I would really appreciate your help, thank you for your offer! 

Could you do that work on the following link?
https://freerouting.crowdin.com/u/projects/1/l/ru

I'll integrate then the new strings into the next release.

andrasfuchs, 2023-10-16:
Hi @rockingdemon, did you have the chance to look into the fixes for the Russian translation?
We are going to have a release in about a week, and it would be great to have better Russian in it ;)

rockingdemon, 2023-10-17:
Hello, @andrasfuchs . Still working on it. Unfortunately it is very inconvenient to use crowding exactly because of string concatenation. I'm currently using this workflow: translate -> compile -> check in running app (context rulez! :-) -> correct, repeat. What's wrong with concatenation. One can't take articles or parts of message as in CleanupWindows bundle and translate them independently. Resulting concatenated messages will still be broken.

Here's what I'm talking about:
[image]

Well, I still can add some improvements (at least menu entries, etc) with static texts. I'll try to upload them this week.

andrasfuchs, 2023-10-18:
@rockingdemon Oh, I see! I agree we should avoid doing concatenation and replace them with templates.

That's a bigger chunk of work, and I don't think I will be able to do it for v1.9, but I'll keep in mind for the future releases.

Don't bother too much with them then now, I appreciate everything else that isn't that much extra work for you.

Thanks again for your contribution!

rockingdemon, 2023-10-20:
@andrasfuchs Hello again! Sorry for delay, had a hard working week. I've ported most of my current translations to crowding. It's not nearly finished, I will continue given enough free time.
Thanks for this project :-)

andrasfuchs, 2023-10-24:
Great work @rockingdemon! Let me know if you feel like it's done for now, so that I can start the integration process from Crowdin.

21km43, 2023-10-30:
I will be proofreading Japanese.

andrasfuchs, 2023-10-30:
This is great, thank you @21km43!

You will need to register and do the proofreading here:
https://freerouting.crowdin.com/u/projects/1

andrasfuchs, 2024-03-04:
According to our stats our userbase mainly prefers English, German, Japanese, Spanish, French, Chinese, Russian, Italian and Portuguese.

Since maintaining different translations isn't trivial and takes time, I'm considering removing support for Hindi, Arabic, 
Bengali and Korean in the next release and I might add Italian.

Please contribute by correcting the auto-trasnlation [on Crowdin](https://freerouting.crowdin.com/).

closed 2024-05-12 by github-actions[bot]