freerouting/freerouting #132: python API?

joamatab, 2022-08-10:
How could we add a python API for freerouting?

it would be great to interface with different tools such as gdsfactory [routers](https://gdsfactory.github.io/gdsfactory/routing.html)

@miho 
@proppy

andrasfuchs, 2022-08-10:
If I understand it correctly you would prefer to have a Python API interface for freerouting, so I'm not sure it helps, but our KiCad integration is done by using Python. You can see the source [here](https://github.com/freerouting/freerouting/tree/master/integrations/KiCad).

joamatab, 2022-08-10:
nice,

ideally we upload it to [PyPi](https://pypi.org) so we can `pip install freerouting`

andrasfuchs, 2022-10-14:
I'm not a Python expert at all, but I image that we would first need to refactor Freerouting to have a separate Java library that have an API that can be called from Python (probably with the help of [Py4J](https://www.py4j.org/index.html)). 
If that worked we could add it as a package to pip with [PyPi](https://pypi.org/).

It doesn't look impossible, but it would be a lot of effort with our current software architecture.

closed 2023-10-12 by github-actions[bot]

andrasfuchs, 2025-04-11:
@abelom Could you also give me repo creation rights as well? I think it would be not too much effort to create separate client libraries in different languages in the future, and they would need new repos each.

abelom, 2025-04-11:
@andrasfuchs just changed your role to 'owner', please validate.