freerouting/freerouting #133: Differential pair routing

Jcwscience, 2022-08-20:
Can freerouting route differential pairs? If not, is this being considered as something that might be added in the future?

andrasfuchs, 2022-08-24:
Could you explain in more detail what would that actually mean regarding the wiring? 
Is my understanding correct that we should then have an option to pre-select connection-pairs, and their wires should follow each other across the board? Does the distance between the two wires matter? Are vias allowed? What if the desired distance is small or bigger then what they have at their beginning and ending footprints?

Could you add some samples with screenshots? Where are these used, and why are they important?

Thank you!

Gigahawk, 2022-11-08:
> Note: not an electrical engineer, this is my best attempt at an explanation

I think a more accurate description would be support for differential impedance controlled lines.
In general, impedance controlled traces are necessary for high speed signals, where any change in impedance along a conductor can cause reflections and distort the signal.

Impedance controlled lines are typically either single-ended (single trace), or differential (a pair of traces)
Single-ended impedance control typically just requires the trace to be some specific width, and potentially some specific length, which is already possible.

Double-ended impedance control requires both traces to be a specific width, as well as a specific distance apart.
Alongside total trace length requirements, usually the length of both traces must be closely matched.
In cases where this is not possible, a "zig-zag" shape is usually added to the shorter trace to make up the difference like this:
[image]

Vias in impedance controlled lines are a fairly complicated topic, in general they should be kept to a minimum.

Note that footprint pads may not match the required width/trace space, in these cases the trace should "fan out" to the correct parameters as soon as possible.
An example (note the pitch of the USB connector to the left is wider than the specified trace space, so shortly after coming out of the USB footprint the traces converge onto the correct trace space)
[image]

closed 2023-10-12 by github-actions[bot]