freerouting/freerouting #137: Pleas create addon plugin

RolandWa, 2022-08-26:
Hi All,

Nice to work on the routing tools. Please add this tools to the KiCad plugin manager using description from
https://dev-docs.kicad.org/en/addons/

Best Regards
Roland

andrasfuchs, 2022-09-06:
Thank you for the suggestion! I agree, it would be great to have [our KiCad plugin](https://github.com/freerouting/freerouting/tree/master/integrations/KiCad) in KiCad's "Plugin and Content Manager".

I already spent a few hours on it, I couldn't make it work then, but I'll get into it again when I get the time.

andrasfuchs, 2022-10-14:
I've just created a merge request for the KiCad addon repo: https://gitlab.com/kicad/addons/metadata/-/merge_requests/122

If everything goes as planned we will have Freerouting listed in KiCad's "Plugin and Content Manager" soon.