freerouting/freerouting #138: Error occurred while opening the dsn file generated by eagle2freerouter

Ansifa, 2022-09-01:
I am using version 1.62 of freerouting, and it prompts an error to open the DSN file:

`E:\Ansifa2021\PCB2021>java.exe -jar freerouting-executable.jar -l en -oit 0.0005 -us hybrid
2022-09-01 10:58:58.759 [main] INFO  Freerouting v1.6.2 (build-date: 2022-07-13)
2022-09-01 10:59:15.364 [AWT-EventQueue-0] INFO  Opening 'ESPMotor_AS5600.dsn'...
2022-09-01 10:59:15.515 [AWT-EventQueue-0] WARN  DsnFile.read_string_scope: closing bracket expected
2022-09-01 10:59:15.519 [AWT-EventQueue-0] WARN  Non-ansi character '-' found at position 187 just after ''Eagle'.
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
2022-09-01 10:59:15.520 [AWT-EventQueue-0] ERROR There was an error while reading DSN file.
java.lang.NullPointerException: Cannot read field "layer_structure" because "routing_board" is null
        at app.freerouting.designforms.specctra.DsnFile.adjust_plane_autoroute_settings(DsnFile.java:77) ~[freerouting-executable.jar:unspecified]
        at app.freerouting.designforms.specctra.DsnFile.read(DsnFile.java:57) ~[freerouting-executable.jar:unspecified]
        at app.freerouting.interactive.BoardHandling.import_design(BoardHandling.java:819) ~[freerouting-executable.jar:unspecified]
        at app.freerouting.gui.BoardFrame.read(BoardFrame.java:230) ~[freerouting-executable.jar:unspecified]
        at app.freerouting.gui.MainApplication.create_board_frame(MainApplication.java:407) ~[freerouting-executable.jar:unspecified]
        at app.freerouting.gui.MainApplication.open_board_design_action(MainApplication.java:479) ~[freerouting-executable.jar:unspecified]
        at app.freerouting.gui.MainApplication.lambda$new$2(MainApplication.java:148) ~[freerouting-executable.jar:unspecified]
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source) ~[?:?]
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source) ~[?:?]
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source) ~[?:?]
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source) ~[?:?]
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source) ~[?:?]
        at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source) ~[?:?]
        at java.awt.Component.processMouseEvent(Unknown Source) ~[?:?]
        at javax.swing.JComponent.processMouseEvent(Unknown Source) ~[?:?]
        at java.awt.Component.processEvent(Unknown Source) ~[?:?]
        at java.awt.Container.processEvent(Unknown Source) ~[?:?]
        at java.awt.Component.dispatchEventImpl(Unknown Source) ~[?:?]
        at java.awt.Container.dispatchEventImpl(Unknown Source) ~[?:?]
        at java.awt.Component.dispatchEvent(Unknown Source) ~[?:?]
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source) ~[?:?]
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source) ~[?:?]
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source) ~[?:?]
        at java.awt.Container.dispatchEventImpl(Unknown Source) ~[?:?]
        at java.awt.Window.dispatchEventImpl(Unknown Source) ~[?:?]
        at java.awt.Component.dispatchEvent(Unknown Source) ~[?:?]
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source) ~[?:?]
        at java.awt.EventQueue$4.run(Unknown Source) ~[?:?]
        at java.awt.EventQueue$4.run(Unknown Source) ~[?:?]
        at java.security.AccessController.doPrivileged(Unknown Source) ~[?:?]
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source) ~[?:?]
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source) ~[?:?]
        at java.awt.EventQueue$5.run(Unknown Source) ~[?:?]
        at java.awt.EventQueue$5.run(Unknown Source) ~[?:?]
        at java.security.AccessController.doPrivileged(Unknown Source) ~[?:?]
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source) ~[?:?]
        at java.awt.EventQueue.dispatchEvent(Unknown Source) ~[?:?]
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source) ~[?:?]
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source) ~[?:?]
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source) ~[?:?]
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source) ~[?:?]
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source) ~[?:?]
        at java.awt.EventDispatchThread.run(Unknown Source) ~[?:?]

`

————————

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
1.62 error, versions below 1.5.0 are normal
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Win11x64 zh_cn, Eagle 9.6.2, java 8u333, eagle2freerouterV6-6.ulp
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
Here is the sample DSN file I use:
[http://121.4.191.35/dsns_and_ulp.zip](url)
* [ ] Did you use any command line arguments?
java.exe -jar freerouting-executable.jar -l en -oit 0.0005 -us hybrid
* [ ] Can you post a screenshot of the problem?
-
* [ ] Can you include the latest, related log entries from the .log file?
-
* [ ] Please list the detailed steps to reproduce the problem
Just open DSN file.
* [ ] What would you expect freerouting to do in this situation?
-
* [ ] Is fixing this issue critical for your work?
-

andrasfuchs, 2022-09-01:
Just quickly looking at the generated DSN, it looks like the 6th line should use " quote characters instead of the current '.

Could you check if it works if you manually change the quotes?

`(host_version 'Eagle V 9.6 - Using eagle2freerouterV6-6.ulp, version 19-4-2014, on 2022/9/1 10:33')`
to
`(host_version "Eagle V 9.6 - Using eagle2freerouterV6-6.ulp, version 19-4-2014, on 2022/9/1 10:33")`

Ansifa, 2022-09-03:
> (host_version "Eagle V 9.6 - Using eagle2freerouterV6-6.ulp, version 19-4-2014, on 2022/9/1 10:33")
Solved! After changing to double quotes, the dsn file can be opened normally, and the problem is solved. I modified the eagle2freerouterV6-6mod.ulp file L519 and L521 into double quotes, and the correct dsn file can be generated. Here is the modified ulp file for those in need.
[http://121.4.191.35/eagle2freerouterV6-6mod.ulp](http://121.4.191.35/eagle2freerouterV6-6mod.ulp)

Thanks a million!

closed 2022-09-03 by Ansifa

Ansifa, 2022-09-03:
Finally learned to upload attachments
[eagle2freerouterV6-6mod.zip](https://github.com/freerouting/freerouting/files/9481811/eagle2freerouterV6-6mod.zip)

andrasfuchs, 2022-09-03:
Excellent, thank you for the modded file!

Do you mind if I integrate it into this repo? It would make sense if we had it here (updated) just like [we have it for KiCad](https://github.com/freerouting/freerouting/tree/master/integrations).

Ansifa, 2022-09-08:
No problem,my pleasure~