freerouting/freerouting #139: Freerouting-1.3.0.jar fails to execute on Mac M1 w/ macOS 12.5.1

geejii, 2022-09-04:
Hi,

Although the Mac binary release of Freerouting is not supported anymore since v1.6.0 [I get the « Freerouting is damaged message »], the jar version is yet supposed to work with Java 17 runtime (and above ?).
I installed the latest Java runtime, V8 Update 341, but freerouting-1.3.0.jar fails to show anything.
I have a Mac mini M1, running macOS 12.5.1/Monterey.

Is that a known issue, or did I do something wrong ?
Please note that I am definitely NOT a Java expert.

Thanks for your help.
Geeji

PS : I am attempting to get autorouting with KiCad 6

andrasfuchs, 2022-09-05:
You are referring to freerouting-1.3.0.jar, which is an old version, are you sure you have the latest? 

Could you download the most recent version and follow the instructions from here:
https://github.com/freerouting/freerouting#running-freerouting-using-java-jre

This method should work on macOS too.

geejii, 2022-09-05:
Thanks a lot for the quick answer :-)

I downloaded the latest « freerouting-1.6.2.jar » on my Finder Desktop and followed the instructions provided at [https://github.com/freerouting/freerouting#running-freerouting-using-java-jre](https://github.com/freerouting/freerouting#running-freerouting-using-java-jre) then [https://adoptium.net/temurin/releases/](https://adoptium.net/temurin/releases/).

There are 2 JDK given there for Mac, I installed the first one : 
« OpenJDK17U-jdk_x64_mac_hotspot_17.0.4.1_1 ».
The installation package completed fine with this success message :
	•	This package has installed:
	•	•	Eclipse Temurin 17.0.4.1+1 to /Library/Java/JavaVirtualMachines/temurin-17.jdk

**I then opened freerouting-1.6.2.jar in Mac Terminal, typing « java -jar ~/Desktop/freerouting-1.6.2.jar », then…
…SUCCESS !! B-)**

Note that simply opening the freerouting-1.6.2.jar file from the Mac Finder, using the default « JavaLauncher » application, does not work at all, while producing no error message.
That was my error since the beginning, but as previously mentioned, I am definitely NOT a Java expert, neither on Mac nor elsewhere.

I suppose the freerouting Java novices like me could benefit from more detailed Mac instructions in your [https://github.com/freerouting/freerouting#running-freerouting-using-java-jre](https://github.com/freerouting/freerouting#running-freerouting-using-java-jre) page, especially about the proper usage of the Mac Terminal :
% java -jar ~/Desktop/freerouting-1.6.2.jar
    instead of on PC :
java.exe -jar ~/Desktop/freerouting-1.6.2.jar

Thanks again for your help, keep the good work.
JJ

andrasfuchs, 2022-09-06:
Excellent, I'm glad it worked! Thank you for your persistence and your notes, I updated the instructions.

closed 2022-09-06 by andrasfuchs