freerouting/freerouting #140: TARGET 3001! supports FreeRouting with "one-click"

HaraldBerlin, 2022-09-08:
* [any/current] What version of freerouting are you using?
* [Windows, Linux/Wine] On what platform did you run it?
* [TARGET 3001!/Specctra/Electra] What type of file are you working with?

We just added a one-click interface to support FreeRouting in our Software TARGET 3001!. The user prepares everything in TARGET and with one click the DSN file is generated and FreeRouting is started. If the SES file is finally generated, TARGET automatically finds it and imports the results. Maybe this could be mentioned on the main page here. Any comments or ideas for improvements on our side are appreciated! [Direct Contact](https://ibfriedrich.com/landingpage/mail-request/en/index.html)

andrasfuchs, 2022-09-09:
Amazing work, thank you!

I'll update the documentation in a few weeks, could you add a few step-by-step instructions here with screenshots?

HaraldBerlin, 2022-10-11:
Hello Andras,

nice to hear from you! We are the people who create the *TARGET 3001!* 
PCB CAD software. We know Alfons Wirtz personally, because he visited 
our office some years ago.

We now have a dialog to start FreeRouting directly:

(Image 1: Open FreeRouting dialog window)

There you can select the signals (=nets) to be routed:

(Image 1: If no special signal is selected, all signals are affected)

Next you can influence the algorithm:

(Image 3: Most users have to install FreeRouting once, some will also 
have to install Java first)

They will get the FreeRouting-Installer from

https://github.com/freerouting/freerouting/releases/download/v1.6.1/freerouting-1.6.1-windows-x64.msi

That path link is always loaded from our server. So we can influence and 
update it.

On the last screen the user can select the layers:

(Image 4: Select layers and their functions)

Normally the user does not have to change the settings and can click 
directly on the [Start] button. So then it is a one-click solution. 
After the creation of the session file SES, Target automatically asks, 
if the results shall be used:

(Image 5: Accept results from SES file)

The tracks and vias are imported immediately into the TARGET project file:

(Image 6: Results are imported)

This is a really easy way of use. And the results look pretty.

Unfortunately we did not get the free-angle routing to life yet. Any 
hints are highly appreciated ;-)

Best regards / Mit freundlichen Grüßen / Meilleures salutations / 
Distinti saluti

      Harald Friedrich

Dipl.-Wirtsch.-Ing. (TU) / Industrial Engineer
CEO / Owner

Ing.-Büro FRIEDRICH

	+49 (0) 6659 919 444 <tel:+496659919444>, Fax: +49 (0) 6659 919 445 
<tel:+496659919445>

	Fuldaer Straße 20, DE-36124 Eichenzell

	***@***.*** ***@***.***>

	www.ibfriedrich.com <https://www.ibfriedrich.com>

andrasfuchs, 2022-10-13:
Thank you @HaraldBerlin for the tutorial steps!

Somehow the screenshots didn't get through, could you please send them to my e-mail: andras.fuchs at gmail.com.

Thank you, I'll update the documentation as soon as I get the screenshots.

I tried to contact Alfons Wirtz a few years ago without any success. Could you pass my contact details to him if you have the chance, please? It would be great to have his oversight of the project, even if he can't/don't want to actively participate.

closed 2022-10-14 by andrasfuchs