freerouting/freerouting #143: Freerouting 1.6.2 does not work on files exported from Eagle 9.5.1 using the latest ULP

Rezer, 2022-09-21:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
1.6.2
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Win 10 x64
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
DSN exported by eagle ULP
[rpi splitter.dsn.txt](https://github.com/freerouting/freerouting/files/9613440/rpi.splitter.dsn.txt)
(Simplest board I had sitting around to demonstrate)
* [ ] Did you use any command line arguments?
No
* [ ] Can you post a screenshot of the problem?
1.6.2 after autorouting:
[image]
1.5.0 after autorouting:
[image]

* [ ] Can you include the latest, related log entries from the .log file?
[freerouting.log](https://github.com/freerouting/freerouting/files/9613446/freerouting.log)

* [ ] Please list the detailed steps to reproduce the problem
Export a DSN file from eagle 9.5.1 using the latest eagle2freerouter.ulp (dated 09/06/2022)
Open the file in freerouting 1.6.2
No unconnected nets are shown and the autorouter does nothing
Repeat the same using freerouting 1.5.0.  Everything works fine.
* [ ] What would you expect freerouting to do in this situation?
Work properly.
* [ ] Is fixing this issue critical for your work?
Sort of!

andrasfuchs, 2022-09-22:
Thank you for posting both the DSN and the log, it helps a lot.
The most likely reason is that the names in the exported DSN file have $ and # characters, so I would start with replacing them and try again.

Todo:
- [ ] Freerouting should generate clear warning messages if this happens, so that needs to be added at least to the log.
- [ ] I'm not sure if the new ULP does anything wrong regarding the component/layer names, it needs to be double-checked.

Rezer, 2022-09-23:
I tried removing those at your suggestion, but it didn't seem to change the behavior at all.  I did notice that there are far less double quotes in the kicad generated files, so I removed all of them below the (parser) directive and it's suddenly working.  I'm not certain which places the quotes are causing trouble, but for some reason the newer versions of freerouting do not like quoted definitions in the dsn file under certain circumstances.  The edited file is below, note that I also had to change "D-" to DMinus instead of D- since the unquoted hyphen was being misinterpreted by the parser.

[rpi splitter.dsn.txt](https://github.com/freerouting/freerouting/files/9630503/rpi.splitter.dsn.txt)

Also there is still a bit of an oddity in that only one autorouted trace shows up until the window is resized or clicked at which time the rest show up, unsure if that's related to the dsn file or a separate issue.

Rezer, 2022-09-23:
I'm unsure of the best way to go about fixing the problem, but I do believe I've at least found the source of the issue.

The problem seems to be in the parsing of the pins list.  When the nets are exported from eagle using the latest ULP they look like this:
    (net "VCC"
      (pins "J2"-"VBUS" "J1"-"VBUS")
    )

However, due to the 1.6.0 change in the Network.java::read_net_pins() from using next_token() to next_string() it does not handle the pins correctly as pairs.  It attempts to parse the next string and split on the hyphen, but next_string() only returns the next _quoted_ string, so the first call returns only J2 and there's no hyphen to split on.  On the next loop next_string() returns -"VBUS" (it does not detect that as a quoted string since it starts with a hyphen), and this pattern repeats with it never storing any component/pin name pairs.

The obvious fix is to make it take the next token if split fails to locate a hyphen, but my java is a bit rusty and I'm uncertain how to handle cases where, for instance, the pin definition is "pin1"-"name with spaces" and the second call to next_string() returns only -"name since it would stop at the first space.  I think reverting to the use of next_token() may be the more straightforward approach, or reevaluating how next_string() determines what constitutes a single string, but I don't know near enough about the design right now to go about mucking around with _that_ one.

Rezer, 2022-09-23:
It occurs to me that it may be simpler to just modify the ULP to output the pin definition as follows:
(pins "J2-VBUS" "J1-VBUS")

However, I'm unsure if that would break backward compatibility with versions before 1.6.0, or if that is even a concern for that matter.

Rezer, 2022-09-23:
Came up with something using next_string_list() instead of next_string() which seems to work well for both quoted and unquoted strings.  Submitted a pull request at https://github.com/freerouting/freerouting/pull/145

Rezer, 2022-09-23:
Aaaaand just found out that smoothieboard.dsn contains pin references specified as C74--, which I didn't think was allowed but apparently is!  This doesn't handle those correctly so I closed that pull request.  To be continued...

Rezer, 2022-09-23:
Alright I think I sorted it out now.  Now it works similar to how it did in 1.5.0, except instead of calling next_token() it uses next_string() but uses the overload where a - can be added to the list of stop characters.  It then throws away the next hyphen and calls next_string() again to get the pin name, which is correctly handled if it is a -.  This is working for eagle files and all the existing test files as far as I can determine...

https://github.com/freerouting/freerouting/pull/146

andrasfuchs, 2022-09-26:
This is great, thank you @Rezer for the thorough investigation and your commitment to fix it! 

I will review your pull request as soon as I can, hopefully later this week!

andrasfuchs, 2022-10-13:
Excellent work, thank you @Rezer!