freerouting/freerouting #150: Window to open file is missing button

denjhang, 2022-10-19:
[image]
[image]
[image]

I'm currently using freerouting v1.6.2, and I found that when I first opened the program, I clicked "Open Your Design" and the file selection window popped up. But the "Open" button in the bottom right corner of the file selection window disappears around 0.5s. After I turned it off and clicked "Open Your Design" again, the "Open" button returned to normal.

denjhang, 2022-10-19:
[image]

andrasfuchs, 2022-10-20:
It's strange, can this be a window sizing issue? The window on the last screenshot is narrower than the others above.

As I tried to reproduce this issue, I could resize my window to a too small size:
[image]

You can fix this by moving the mouse cursor to the bottom right corner of the window and resize it.

The fix should be that a minimum width and height must be set to this (and I think all other) windows.

denjhang, 2022-10-28:
Although I adjusted the window to temporarily solve the problem, the enter key did not work. When I close the window and open it for the second time, everything works fine.

andrasfuchs, 2023-02-15:
It got fixed in version 1.7.0, so I close this now.

closed 2023-02-15 by andrasfuchs