freerouting/freerouting #151: It doesn't find the FreeRoute

tormyvancool, 2022-11-01:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using? **v1.4.4 .EXE ando also a previous one in .JAR**
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? **Windows 11 Pro 64bit**
* [ ] What type of file are you working with? Can you attach the actual (DSN) file? **PCB files out of KiCAD (6.08 and 6.09).  No I can't due NDA**
* [ ] Did you use any command line arguments? No
* [ ] Can you post a screenshot of the problem? **yes here below**
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation? Just be able to indicate where my Freerotue is located, since it's not on C driver at all.
* [ ] Is fixing this issue critical for your work? **yes it is**

From inside the PCB of KiCAD I try to launch Freeroute clicking oth the button located on the main toolbar but I got this screenshot
So, I have to run Freeroute manually as I usually do. But if your plugin was working I could using from inside KiCAD directly

[image]

andrasfuchs, 2022-11-01:
Yes, looking at your screenshot the mixing of `\` and `/` characters in the path doesn't look right.
First I thought that we need to change [the plugin script](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/kicad-freerouting/plugins/plugin.py) to use the correct one depending on the platform, but as far as I can see we use `os.path.join` which should take care of this. 
It needs further investigation.

tormyvancool, 2022-11-24:
The script is in facts unusable. I had to uninstall your package. Sorry for this.
When ther ewill be a working/stable one I will be happy to install it

andrasfuchs, 2023-01-03:
Sorry, I understand your frustration, the error message you got isn't really helpful.

The reason was that you had Java 8 installed, but Freerouting needs Java 17.

tormyvancool, 2023-01-03:
> 

Well normally I perform updates each time the system requires. So, if Java17 is required but I have Java8 it means i didn't get any updates from the 8 on ... where do I find it so I iwll see if ever the issue is popping up again?

andrasfuchs, 2023-01-03:
Hmm, that's strange. Is it possible the you have both Java JRE 8 and 17 installed? In that case I can imagine that the default Java path is set to 8, and not 17. I guess KiCad tries to run freerouting by calling the default Java JRE (and not the latest one installed). Otherwise I think it should work!

In any case, you can download Java 17 from here: https://adoptium.net/temurin/releases/

tormyvancool, 2023-01-03:
> Hmm, that's strange. Is it possible the you have both Java JRE 8 and 17 installed? In that case I can imagine that the default Java path is set to 8, and not 17. I guess KiCad tries to run [...]

Well not I have only the 8:
[image]

tormyvancool, 2023-01-03:
Oh damn ... 
[image]

andrasfuchs, 2023-01-03:
I made the publication of the new version an hour ago, so version 1.6.5 should be there in the KiCad Plugin Manager in a few hours. I hope that will work, but let me know if it still doesn't!

Sorry for the problems, and thank you for your reports!

tormyvancool, 2023-01-03:
> I made the publication of the new version an hour ago, so version 1.6.5 should be there in the KiCad Plugin Manager in a few hours. I hope that will work, but let me know if it still doesn't!
> 
> Sorry for the problems, and thank you for your reports!

I installed this https://www.oracle.com/java/technologies/downloads/#java17 but your package still not available, for the time being

andrasfuchs, 2023-01-03:
It should be there soon, but it's out of my control when.

Until then you can download it [from here](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/kicad-freerouting-1.6.5.zip?raw=true) and choose `Install from File...` in the KiCad Plugin Manager, if you like to try it.

tormyvancool, 2023-01-04:
> It should be there soon, but it's out of my control when.
> 
> Until then you can download it [from here](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/kicad-freerouting-1.6.5.zip?raw=true) and choose `Install from File...` in the KiCad Plugin Manager, if you like to try it.

I was constrained to install it from file ... since it's not yet available on the plugin manager.

now it damn works :-)  BUT:

1 . Into the toolbar there is not a distinctive icon
[image]

andrasfuchs, 2023-01-06:
I'm glad it finally works! :)

Yes, good point, we'd need to replace that icon!

tormyvancool, 2023-01-06:
> I'm glad it finally works! :)
> 
> Yes, good point, we'd need to replace that icon!

I think they must also to improve their policies for the updates. If anyone tries to update the package by the package manager, even after days, it returns error.

[image]

andrasfuchs, 2023-01-06:
That is odd. Can you open the URL directly in your browser?
https://github.com/freerouting/freerouting/raw/master/integrations/KiCad/kicad-freerouting-1.6.5.zip

andrasfuchs, 2023-01-06:
I added the Java version check and fixed the toolbar icon for the KiCad plugin.

Here is [my merge request](https://gitlab.com/kicad/addons/metadata/-/merge_requests/151), so if/when they merge and publish it, these will be fixed.

tormyvancool, 2023-01-06:
> That is odd. Can you open the URL directly in your browser? https://github.com/freerouting/freerouting/raw/master/integrations/KiCad/kicad-freerouting-1.6.5.zip

Yes I could do it in the broswer without issue

tormyvancool, 2023-01-06:
> I added the Java version check and fixed the toolbar icon for the KiCad plugin.
> 
> Here is [my merge request](https://gitlab.com/kicad/addons/metadata/-/merge_requests/151), so if/when they merge and publish it, these will be fixed.

I re-installed it form the link you gave me right above, but the icon still the generic one
[image]

andrasfuchs, 2023-01-08:
The new, fixed version got published to the KiCad Plugin Manager only an hour ago. Could you try 1.6.6?

[image]

[image]

[image]

tormyvancool, 2023-01-09:
This is what I get:

[image]

andrasfuchs, 2023-01-09:
I tried installing it from multiple countries (by using VPN) using KiCad 6.0.5, and they all worked.

It looks like it's out of my control at this point, it might be a bug in KiCad 6.0.10 or a network configuration you have.

If anyone else reads this, please comment below if you can or can't download the archive (like @tormyvancool) so that we can continue the investigation.

tormyvancool, 2023-01-09:
> I tried installing it from multiple countries (by using VPN) using KiCad 6.0.5, and they all worked.
>
> It looks like it's out of my control at this point, it might be a bug in KiCad 6.0.10 or a [...]

I opened a thread on the forum, referring to this issue. Hopefully they can check and come up with a solution.

https://forum.kicad.info/t/plugin-manager-issue-downloading-plugins/39460/15

rei-vilo, 2023-01-16:
Same problem here, `OpenJDK 64-Bit Server VM Temurin-17.0.5+8` installed but the plug-in complains about the missing Java 17 JRE.

[image]

```
$ java --version
openjdk 17.0.5 2022-10-18
OpenJDK Runtime Environment Temurin-17.0.5+8 (build 17.0.5+8)
OpenJDK 64-Bit Server VM Temurin-17.0.5+8 (build 17.0.5+8, mixed mode, sharing)
```

Kicad is release `6.0.10`

```
Application: KiCad

Version: 6.0.10, release build

Libraries:
	wxWidgets 3.2.1
	libcurl/7.86.0-DEV OpenSSL/3.0.7 zlib/1.2.12 nghttp2/1.51.0

Platform: Linux 6.0.0-0.deb11.2-amd64 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Jan 14 2023 11:12:49
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.86.0-DEV
	ngspice: 38
	Compiler: GCC 12.1.0 with C++ ABI 1017

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

andrasfuchs, 2023-01-17:
@rei-vilo Thanks for the report, I'll need to look into this, but I can't in the next few weeks.

Is the java executable part of the PATH variable, so can it be executed from anywhere on your system? If yes, then I suspect that it's a Linux specific issue that I need to fix later.

Until then you can try version 1.6.5, they are basically the same except the Java version checking during the plugin installation.

rei-vilo, 2023-01-17:
Thank you for your prompt answer! Yes, the path to `java` is included in the `PATH` variable.

```
$ which java
/usr/bin/java
$ echo $PATH
/home/USER/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/USER/.local/bin
```

There is no hurry: I am using FreeRouting manually. Good luck!

tormyvancool, 2023-01-29:
With version 6.0.11 it's solved ... I hope still solved with the 7 ...

andrasfuchs, 2023-02-07:
@rei-vilo Could you confirm that it works for you as well with KiCad 6.0.11?

If it still not doesn't, could you run the `java -version` command, and tell me what the exact output that you have?

rei-vilo, 2023-02-07:
Thank you for coming back. Kicad is now `Version: 6.0.11, release build` but the plug-in still complains about Java 17.
 
```
$ java -version
openjdk version "17.0.6" 2023-01-17
OpenJDK Runtime Environment (build 17.0.6+10-Debian-1deb11u1)
OpenJDK 64-Bit Server VM (build 17.0.6+10-Debian-1deb11u1, mixed mode, sharing)

$ which java
/usr/bin/java

$ echo $PATH
/home/USER/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/USER/.local/bin

/home/USER/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/USER/.local/bin
```

andrasfuchs, 2023-02-07:
@rei-vilo I modified the regular expression used for version detection and I have just uploaded a file that can be run separately to check the Java version. Could you help me by running this small python script on your machine?

Here is [the checkjava.py](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/checkjava.py), you can run it by the `python -c "import checkjava; checkjava.get_java_version()"` command if you have Python 3 installed.

Please send me back the exact output you have, thank you!

rei-vilo, 2023-02-08:
Here is the result:

```
$ python -c "import checkjava; checkjava.get_java_version()"
openjdk version "17.0.6" 2023-01-17
17.0.6
```

andrasfuchs, 2023-02-08:
Great, thank you!

rei-vilo, 2023-02-13:
I tested the new release `1.7.0` on Kicad `7.0`. 

I am using [Kicad installed as FlatPak](https://www.kicad.org/download/flatpak/) on Debian 11.

Alas, the same error message appears. 

1. I edited line

https://github.com/freerouting/freerouting/blob/e7ab5f9064b9453103693a84201f1e0cfc30a83d/integrations/KiCad/kicad-freerouting/plugins/plugin.py#L262

to
 
```
        return "17.0.6"
```

2. I cleaned the `__pycache__` and called the plug-in again.

[image]

3. I used FlatSeal to edit the authorization and allow access to system files, and executables. 

4. I also listed `/usr/bin/java` in specific files.

None of the actions below solved the issue. 

It looks like Kicad installed with FlatPak and the plug-in do not have access to the system.

rei-vilo, 2023-02-13:
Instead of using [Kicad installed as FlatPak](https://www.kicad.org/download/flatpak/), I took the [back-port version for Debian](https://www.kicad.org/download/debian/).

Everything works fine now: I am calling FreeRouting from the plug-in menu of Kicad PCB.

andrasfuchs, 2023-02-15:
Interesting! We might have a new problem with KiCad 7 or Flatpak?

There are a few more issues related to KiCad plugin (#169 #168 and #170), let's get them fixed first, so if it's stable with KiCad 6, I can start testing with KiCad 7.

Thanks for the details @rei-vilo!

rei-vilo, 2023-02-16:
You're welcome! 

> Interesting! We might have a new problem with KiCad 7 or Flatpak?

As applications bundled by FlatPak run in a sandbox, I suspect this is the cause of the issue, even though I had authorised the access to the system files and Java in particular.

It might be worth asking FlatPak and Kicad about how the plug-ins of an application are managed.

rei-vilo, 2023-02-16:
You're welcome! 

> Interesting! We might have a new problem with KiCad 7 or Flatpak?

As applications bundled by FlatPak run in a sandbox, I suspect this is the cause of the issue, even though I had authorised access to the system files and Java in particular.  

It might be worth asking FlatPak and Kicad about how the plug-ins of an application are managed.

retiredfeline, 2023-05-12:
From what I understand of flatpaks, and having built one myself for another project, as well as some searching of the relevant *stacks, it is indeed the sandbox that prevents execution of Java programs using the system's Java environment.

Unfortunately that's not the whole story. There is an openjdk extension to the freedesktop runtime (one of three major runtimes flatpaks can be built for) but it's for the **Sdk**! That is to say, it's only useful to the person building the KiCad flatpak to add Java execution. It's not possible to add a Java "runtime" to an existing flatpak installation.

So, sorry, the answer is don't use the KiCad flatpak if you want to use freeRouting ahd hence Java. Use the native packages for your distro.

andrasfuchs, 2023-05-12:
Thank you for the insight @retiredfeline!
I have no experience with Flatpak, so it's probably a noob questions, but what do you think, could KiCad decide to allow Java execution in their Flatpak container if they changed their Flatpak settings/setup?

Btw, what the most recent plugin installer script does is that if Java can't be found or it's not accessible (because of Flatpak) it tries to install JRE 17 into the system temp directory on the fly and run the Freerouting app using that installation. 

I don't know yet, if it works, we'll see when the community provides some feedback after the next release.

retiredfeline, 2023-05-12:
Hi, this is what would have to happen:

- The volunteer building the flatpak (not the KiCad project or the developers) would have to add the openjdk extension to their Sdk for building the flatpak. BTW, the openjdk extension is for the freedesktop Sdk, and I don't know if the GNOME Sdk supports the openjdk extension.
- The flatpak gets bloated because it's now carrying an openjdk runtime environment inside the flatpak so that the java interpreter can run inside the sandbox.

So I don't know if the flatpak packager would agree to that.

retiredfeline, 2023-05-14:
A long term development that might help is to enhance freeRouting to be able to be run as a service that accepts jobs and returns the result over the network. This would mean running headless and having no interactive control over the job. Something would have to be put in place to deal with jobs that won't terminate or take too long. But it would enable the service to be shared across many users without them having to install Java. It could be run on a powerful compute server.

This is just speculation and probably deserves a separate discussion, if at all.

andrasfuchs, 2023-05-14:
Yes, I agree. I had similar plans, but I had to realize that there is not enough community support at the moment to do the development. I'm the only maintainer, I work for free when I have a few days and it's just too big of a change without help.

I almost got a small team of Java developers from the University of Copenhagen to work on it as their assignment, so if any of you have connection to universities, telling them about Freerouting would be a great way to show your support! ;)

Here is the e-mail I wrote to them in 2022, feel free to use any part of it: https://github.com/freerouting/freerouting/discussions/193

retiredfeline, 2023-05-14:
If it's possible to make freeRouting run headless and unattended from the command line, I'd be willing to work on a Python server to accept input over the network, run freeRouting, and return the output over the network. This can then be converted to Java if desired later. Essentially doing as a server what a user does from the KiCad plugin.

andrasfuchs, 2023-05-15:
Sounds great!

We only have [a partial headless implementation](https://github.com/freerouting/freerouting/blob/master/src/main/java/app/freerouting/interactive/BoardHandlingHeadless.java), but scripting is still possible with the [command line arguments](https://github.com/freerouting/freerouting#using-the-command-line-arguments). If you set both the input (`-de`) and the output (`-do`) files Freerouting automatically loads the input file, does the routing, saves the output file and exits. Logs are written to both the console+stderr and into log files.

Is this enough, or you would need a truely headless execution? 
If you are willing to work on this, please open a new issue, so we can discuss the further details there.

Thank you for considering it!

closed 2023-10-12 by github-actions[bot]