freerouting/freerouting #153: Open dsn is stuck loading

timrosskamp, 2022-11-11:
I can't open any .dsn files exorted by KiCAD. I've tried this on Windows 10 64-bit and MacOS 64-bit. Both times I'm stuck on loading and get a heap space error after a while.

- Version v1.6.2
- MacOS 64-bit
- [wavefolder.dsn.zip](https://github.com/freerouting/freerouting/files/9991073/wavefolder.dsn.zip)

```
╰─➤  java -jar freerouting.jar -de wavefolder.dsn 
2022-11-11 15:52:17.982 [main] INFO  Freerouting v1.6.2 (build-date: 2022-07-13)
2022-11-11 15:52:17.989 [main] INFO  Opening 'wavefolder.dsn'...
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
2022-11-11 15:54:14.291 [main] ERROR Java heap space
java.lang.OutOfMemoryError: Java heap space
	at java.lang.StringBuffer.toString(StringBuffer.java:715) ~[?:?]
	at app.freerouting.designforms.specctra.SpecctraDsnFileReader.next_string(SpecctraDsnFileReader.java:1732) ~[freerouting.jar:unspecified]
	at app.freerouting.designforms.specctra.SpecctraDsnFileReader.next_string_list(SpecctraDsnFileReader.java:1743) ~[freerouting.jar:unspecified]
	at app.freerouting.designforms.specctra.Rule.read_clearance_rule(Rule.java:225) ~[freerouting.jar:unspecified]
	at app.freerouting.designforms.specctra.Rule.read_scope(Rule.java:39) ~[freerouting.jar:unspecified]
	at app.freerouting.designforms.specctra.Structure.read_scope(Structure.java:834) ~[freerouting.jar:unspecified]
	at app.freerouting.designforms.specctra.ScopeKeyword.read_scope(ScopeKeyword.java:62) ~[freerouting.jar:unspecified]
	at app.freerouting.designforms.specctra.DsnFile.read(DsnFile.java:51) ~[freerouting.jar:unspecified]
	at app.freerouting.interactive.BoardHandling.import_design(BoardHandling.java:819) ~[freerouting.jar:unspecified]
	at app.freerouting.gui.BoardFrame.read(BoardFrame.java:230) ~[freerouting.jar:unspecified]
	at app.freerouting.gui.MainApplication.create_board_frame(MainApplication.java:407) ~[freerouting.jar:unspecified]
	at app.freerouting.gui.MainApplication.main(MainApplication.java:261) ~[freerouting.jar:unspecified]
^C%   
```

timrosskamp, 2022-11-11:
I've just tried to open the file with version 1.5.0 which is working.

eltorio, 2022-12-05:
quite same here:
```sh
me@ 1FORefit % java -jar /Applications/KiCad/Pcbnew.app/Contents/MacOS/freerouting-1.6.2.jar -de 1FORefit.dsn 
2022-12-05 16:10:37.503 [main] INFO  Freerouting v1.6.2 (build-date: 2022-07-13)
2022-12-05 16:10:37.523 [main] INFO  Opening '1FORefit.dsn'...
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
2022-12-05 16:15:21.241 [main] ERROR Java heap space
java.lang.OutOfMemoryError: Java heap space
	at java.util.Arrays.copyOfRange(Unknown Source) ~[?:?]
	at java.lang.String.<init>(Unknown Source) ~[?:?]
	at java.lang.StringBuffer.toString(Unknown Source) ~[?:?]
	at app.freerouting.designforms.specctra.SpecctraDsnFileReader.next_string(SpecctraDsnFileReader.java:1732) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.designforms.specctra.SpecctraDsnFileReader.next_string_list(SpecctraDsnFileReader.java:1743) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.designforms.specctra.Rule.read_clearance_rule(Rule.java:225) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.designforms.specctra.Rule.read_scope(Rule.java:39) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.designforms.specctra.Structure.read_scope(Structure.java:834) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.designforms.specctra.ScopeKeyword.read_scope(ScopeKeyword.java:62) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.designforms.specctra.DsnFile.read(DsnFile.java:51) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.interactive.BoardHandling.import_design(BoardHandling.java:819) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.gui.BoardFrame.read(BoardFrame.java:230) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.gui.MainApplication.create_board_frame(MainApplication.java:407) ~[freerouting-1.6.2.jar:unspecified]
	at app.freerouting.gui.MainApplication.main(MainApplication.java:261) ~[freerouting-1.6.2.jar:unspecified]
```
Mac OS 13.1
1FORefit % java -version
openjdk version "19.0.1" 2022-10-18
OpenJDK Runtime Environment Temurin-19.0.1+10 (build 19.0.1+10)
OpenJDK 64-Bit Server VM Temurin-19.0.1+10 (build 19.0.1+10, mixed mode, sharing)

andrasfuchs, 2023-01-03:
I tested the DSN file with the snapshot version and it seems to be working fine, so you can download that version[ from the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT) or wait for the next release.

mik13ST, 2023-01-15:
I have the same problem and I managed to simplify my KiCad circuit to one resistor on a board which is enough to cause this issue in 1.6.5.

[minimal-freerouting-crasher.dsn.zip](https://github.com/freerouting/freerouting/files/10419957/minimal-freerouting-crasher.zip)

1.5.0 works fine.
Snapshots "20230109_151600" and "20230109_151900" don't work.
1.6.5 doesn't work.

Since a change in code of this project influences this and we know a good and bad version, one could do a git bisect to find which commit did this. I will try to setup building from source and then I will try the bisect.

mik13ST, 2023-01-15:
I was setting up my build environment on Windows and messing with my JRE and JDK versions and I noticed something. This problem only happens on JRE/JDK 19, but not 18 or 17. Since 17 is LTS and 19 is not, addressing this is not urgent if JRE 17 is recommended to run Freerouting which is indeed the case (see https://github.com/freerouting/freerouting#running-freerouting-using-java-jre).

The solution for others with this problem: use JRE 17.

mik13ST, 2023-01-15:
I ran the bisect anyways and found https://github.com/freerouting/freerouting/commit/249e9555e8208e3c57e502c6bf8922469495d119 to be the first commit where my testing DSN file fails to read with "unable to read .dsn-file" message, but it doesn't hang. Commit https://github.com/freerouting/freerouting/commit/b0fa3971e383832f164cb83b00f7aed234d36c1f makes it worse and causes it to hang on "loading .dsn file..." instead of the nice message. If left hanging for too long, it may run out of memory and that must be what people are reporting. In my case, when loading is stuck, JRE uses 100% CPU and about 2GB of RAM.

Again, this is only a problem on JRE 19, not 18 or 17.

I don't know if it makes sense to address these issues since 19 is not LTS and the next LTS version might behave differently again. Also, I don't know the code. I just wanted to clear up this issue.

I skipped setting up build environment on Windows by using https://gitpod.io. I gave it this repo URL, changed Java version to 17 using https://www.gitpod.io/docs/introduction/languages/java#updating-java-version and `./gradlew` was working and building. Very convenient.

andrasfuchs, 2023-02-15:
It got fixed in version 1.7.0, so I close this now.

closed 2023-02-15 by andrasfuchs