freerouting/freerouting #155: Autorouter causing a lot of clearance violations with pads

ProxyPlayerHD, 2022-11-15:
* What version of freerouting are you using?
v1.6.2 (build 2022-07-13)

* On what platform did you run it?
Windows 10

* What type of file are you working with?
Spectra DNS exported from KiCad

* Did you use any command line arguments?
"-l en"

* Can you post a screenshot of the problem?
[image]

* Please list the detailed steps to reproduce the problem
Click "Autorouter", wait for it to finish

* What would you expect freerouting to do in this situation?
not this

* Is fixing this issue critical for your work?
nah i'm just a hobbyist, but having the autorouter not work kinda defeats the purpose of it...

DNS and Log are in this Zip File: 
[FR Stuff.zip](https://github.com/freerouting/freerouting/files/10015149/FR.Stuff.zip)

hope this is enough info to find out what is happening

DLehenbauer, 2023-01-05:
@ProxyPlayerHD - I'm curious if you tried '-mt 1' to disable multithreading?

Anecdotally, I've found that this problem is exacerbated on fast machines with many cores.

DomoticX, 2023-01-31:
I can confirm this issue, using version 1.6.2 and 1.6.5....when using all cores (32 on R9-7950X), some tracks just go trough the pads, using the '-mt 1' command solves this problem for now.

ProxyPlayerHD, 2023-02-07:
i could've sworn i already commented... hmm.
anyways yes, using `-mt 1` avoids the issue. of course that isn't a solution, but it works for now.

andrasfuchs, 2023-02-07:
Thank you guys for the reports, it would be important to fix this in the next release.

I agree that if I can't allocate the time for the bugfix in the next week or so, it would be better to have a new release with multithreading disabled by default.

andrasfuchs, 2023-02-08:
Alright, so I spent a few hours on this issue, and although I cleaned up the multithreading task scheduler code, I don't think it makes sense to use it. The multi-threader tries to optimize multiple items on the board at the same time, but if they are related it can't guarantee the clearances. I think it is much better to keep it single threaded even if this means that the optimization process takes a little longer.

I will keep the `-mt` option for now just in case we want to run experiments, but its default value will be 1, so the route optimizer will be single threaded.

andrasfuchs, 2023-02-15:
It got fixed in version 1.7.0, so I close this now.

closed 2023-02-15 by andrasfuchs