freerouting/freerouting #156: Enhancment Suggestion: "Bend Cost" to aid in autorouting for single sided fabrication.

sleemanj, 2022-11-30:
When routing a board for single-sided fabrication, especially DIY boards it is useful to have what will in reality be the non-copper side only have purely straight line routes which are implemented during fabrication by a simple straight solid wire, or for short ones by an SMD 0-ohm link on the copper side.

It is **almost** possible to do it already in Freerouting by setting the cost for "against preferred direction" on layer 2 to a very very high number (10000.0) and then it will make basically all routes on layer 2 straight and in the preferred direction.  However that will mean that layer 2 ONLY has all auto routes vertical, or all auto routes horizontal.  Ideally it should allow to have both horizontal and vertical auto routes on that layer, as long as each route individually is a straight line.

I propose the idea that would I think be able to accomplish this and more is to add a "bend cost" parameter for the "Trace costs on layer" autoroute parameters, such that if a route bends that increases the cost.

sleemanj, 2023-10-04:
Comment.

sleemanj, 2023-12-05:
Comment ---- On Tue,05 Dec 2023 14:45:43 +1300  ***@***.***  wrote ----
This issue is stale because it has been inactive for 60 days. Remove stale label or comment or this will be closed in 7 days.

sleemanj, 2024-02-07:
comment

andrasfuchs, 2024-03-04:
@sleemanj Thanks for the suggestion. Is it similar to issue #278?

If not, could you explain it the difference through an example?

Thank you!

sleemanj, 2024-03-04:
In general my idea is simply "changing direction could have a configurable cost".It sounds like the other ticket is the more or less the same but is making the trace after the direction change more expensive rather than the direction change itself.  I imagine either approach would probably have the same effect.Message ID: ***@***.***>

sleemanj, 2024-05-04:
comment

sleemanj, 2024-09-02:
comment

sleemanj, 2025-01-01:
Comment

sleemanj, 2025-05-03:
Comment

Message ID: ***@***.***>

sleemanj, 2025-09-01:
Comment

sleemanj, 2025-12-31:
Comment

sleemanj, 2026-05-01:
Comment