freerouting/freerouting #160: Plugin doesn't work in Kicad 6.0.9

drelephant, 2023-01-06:
* freerouting v1.6.5
* Windows 10 64bit

Following the instructions from [here](https://github.com/freerouting/freerouting/tree/master/integrations/KiCad).

Opened Kicad, went to Tools-External Plugins-Open Plugin Directory

Saved `kicad-freerouting-1.6.5.zip` in there (I also tried extracting to that folder).

BTW there are two copies of the identical file saved in "freerouting/integrations/KiCad/" - one called [kicad-freerouting-1.6.5.zip](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/kicad-freerouting-1.6.5.zip) and the other called [kicad-freerouting.zip](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/kicad-freerouting.zip).

Went to Tools-External Plugins-Refresh Plugins.

The Kicad screen flashed/refreshed quickly, but I don't see any Freerouting option anywhere.

I can run the jar inside the plugins/jar folder manually and it brings up the window successfully.

* [ ] What would you expect freerouting to do in this situation?

Show up under Plugins in Kicad.

* [ ] Is fixing this issue critical for your work?

No, it's simply a minor annoyance. Thanks for the great free software!

andrasfuchs, 2023-01-06:
Could you try to install it from the KiCad Plugin and Content Manager? I have KiCad 6.0.5, and it works fine for me. You might need to remove the previous manual installation first.

[image]

Having identical zips is intentional: we need to have one with the version number in filename because of the KiCad Plugin Manager, but I also wanted to keep one with a more general filename for the always-up-to-date version, so others can have a permanent link to it.

I should update the instructions, it's probably outdated, thank you for reporting!

drelephant, 2023-01-06:
That worked perfectly, thanks!!

What you really need to get rid of is the instructions at https://freerouting.org/freerouting/using-with-kicad - that was the first google result that came up for "freerouting kicad".

I did that last night and it was a terrible experience. I had to install the gigantic "LayoutEditor" just to get a completely outdated freerouting.jar which had a horrible file browser UI and even came up with a broken blank screen with no buttons because my monitor was not at 100% dpi, something [I discovered after googling](https://forum.kicad.info/t/freerouting-displays-strangely-on-windows-10-solved/17783) for the problem.

The new version seems so much better, and the Kicad integration seems great so far.

andrasfuchs, 2023-01-06:
I'm really sorry you run into those instructions! The problem is that I/we don't control the freerouting.org domain, and although I tried to contact them earlier to get in sync and update the site I wasn't successful. But now that you mentioned it, I'll try again, you never know.

I'll start with updating [our own documentation](https://github.com/freerouting/freerouting/tree/master/integrations/KiCad), because that's outdated as well.

Thanks again!

andrasfuchs, 2023-01-06:
So, our documentation is updated, and I also the first few search results on Google:
1. [freerouting.org](https://freerouting.org/freerouting/using-with-kicad) owned by [Jürgen Thies](https://www.linkedin.com/in/j%C3%BCrgen-thies-67743111/). We contacted him in March, 2020 without success, I tried again today.
2. This repo 👍 
3. [freerouting.mihosoft.org](https://freerouting.mihosoft.eu/) owned by Michael Hoffer - although his website is not updated, he is a contributor here, so it's likely that he is willing to do the updates. I have just dropped him a line.
4. [TechExplorations article](https://techexplorations.com/blog/kicad/blog-kicad-5-how-to-use-the-freerouting-autorouter/), I could comment below the article with the updated instructions.
5. [Relevant KiCad forum thread](https://forum.kicad.info/t/maybe-a-bug-between-kicad-and-freerouting-jar/28206), but unfortunatelly it's locked, I can't post there.
6. [Wikibooks page](https://en.wikibooks.org/wiki/Kicad/Autorouter), I might be able to update this, it depends on the admins, we'll see.
7. [Reddit post](https://www.reddit.com/r/KiCad/comments/s1k5a1/autoroute_in_kicad_6/), I added my comment with the link to the intructions.

closed 2023-01-06 by andrasfuchs