freerouting/freerouting #161: 【UI】incorrect freerouting icon

EasyEDA2021, 2023-01-13:
not a important issue
[image]
the icon not freerouting icon, it shows java icon

the open dialog is correct
[image]

but the other dialogs are not correct
[image]

andrasfuchs, 2023-02-15:
It got fixed in version 1.7.0, so I close this now.

EasyEDA2021, 2023-02-16:
> It got fixed in version 1.7.0, so I close this now.

hi，I tested it, still have windows no icon, the task bar no icon too
[image]
[image]
[image]
[image]

maybe it was because of overiding old version to install new version(old version cache) ?

andrasfuchs, 2023-02-17:
Ah, you are right, I replaced the icon on the windows, but not for the application itself.

Thanks, I'll reopen this, and try to fix next time.

andrasfuchs, 2023-05-04:
It will be fixed in the next release.

[image]

closed 2023-05-22 by andrasfuchs