freerouting/freerouting #162: Race condition in opt_route_pass thread pool wait.

chrisbmr, 2023-01-15:
https://github.com/freerouting/freerouting/blob/67b791e53835a4dce8529899f3a3e2495551c829/src/main/java/app/freerouting/autoroute/BatchOptRouteMT.java#L283

All active threads may have already counted down the latch before exiting (getActiveCount() changing) and setting the latch back to 1.

    t0: Schedule t1,t2.
    t0: pool.getActiveCount() >= thread_pool_size
    t1,t2: task_completion_signal.countDown()
    t0: prepare_task_completion_signal() sets latch from 0 to 1.
    t1,t2: exit
    Waiting for 3 minutes because the tasks have already exited and nobody will signal the latch.

andrasfuchs, 2023-02-08:
Thank you for your report @chrisbmr. I simplified the scheduler and updated the task completion check, so hopefully it will work properly. Keep in mind that multi-threading will be disabled in the future by default because of another issue (#155).

andrasfuchs, 2023-02-15:
It got fixed in version 1.7.0, so I close this now.

closed 2023-02-15 by andrasfuchs