freerouting/freerouting #170: Java version check fails

dhoepfl, 2023-02-15:
To make our work easier, please provide as many details as you can:
* [x] What version of freerouting are you using?
1.7.0
* [x] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Windows/64
* [x] What EDA do you use? KiCad, EAGLE, other?
KiCad
* [x] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
Irrelevant
* [x] Did you use any command line arguments?
No
* [x] Can you post a screenshot of the problem?
Irrelevant
* [x] Can you include the latest, related log entries from the .log file?
Irrelevant
* [x] Please list the detailed steps to reproduce the problem
See below
* [x] What would you expect freerouting to do in this situation?
Respect configuration/Allow override.
* [x] Is fixing this issue critical for your work?
Workaround deployed.

The KiCad plugin for freerouting allows to configure the java path in its settings file. The java version test however ignores the configured path [see this line of code](https://github.com/freerouting/freerouting/blob/e7ab5f9064b9453103693a84201f1e0cfc30a83d/integrations/KiCad/kicad-freerouting/plugins/plugin.py#L256).

Thus the version check might use a different java than the actual call uses. This leads to failed check if there is no java in the path or if the java in the path is outdated.

Further more, the check fails if the java executable produces an unexpected output (e.g. javaw.exe from the Corretto JDK does not print anything for -version). Maybe it would be better to start a tiny Java program that reports the Java version in a defined format.

andrasfuchs, 2023-02-15:
You are right, it's a bug and it should use the configured Java path, thank you for the report.

Regarding to Corretto, I got this answer from Bing:
[image]

Can you confirm that this is indeed the output of the `java -version` command?

dhoepfl, 2023-02-16:
That’s correct, the output is similar:

```
C:\Users\xxxxxx>C:\…\jdk18.0.2_9\bin\java.exe -version
openjdk version "18.0.2" 2022-07-19
OpenJDK Runtime Environment Corretto-18.0.2.9.1 (build 18.0.2+9-FR)
OpenJDK 64-Bit Server VM Corretto-18.0.2.9.1 (build 18.0.2+9-FR, mixed mode, sharing)
```

My fault was that I set `java_path` to use `…\javaw.exe` (out of habit to suppress the console window) which does not report anything for `-version`:

```
C:\Users\xxxxxx>C:\…\jdk18.0.2_9\bin\javaw.exe -version

C:\Users\xxxxxx>
```

That’s why I suggested to run a tiny Java program prints the version in a defined form. Something like `System.out.println(Runtime.version().feature());`.

andrasfuchs, 2023-02-16:
I have just applied a few changes to the java version checker, but if it doesn't work then your suggestion could be another option as well.

Thank you, I'll publish a new version in the following days.

dhoepfl, 2023-02-16:
Thanks a lot!

andrasfuchs, 2023-03-01:
I modified the Java check again in version 1.7.1 of the KiCad plugin. Is it still broken?

Theoretically you can fix this by setting the path to Java manually. I think you should have a `plugin.ini` somewhere and that has a `path` variable that you can change to your current `java` executable (eg. `/usr/bin/java`).

Mine (with KiCad 6 on Windows 11) is at `...\Documents\KiCad\6.0\3rdparty\plugins\app_freerouting_kicad-plugin\`.
[image]

kaosbeat, 2023-03-02:
I found two
`~/.var/app/org.kicad.KiCad/data/kicad/7.0/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.ini`
and
`~/.local/share/kicad/7.0/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.ini`
I edited both to `/usr/bin/java` from `java`
no solution. I cannot update to 1.7.1 though, I have to wait until it is pushed to the stable repo? or can I add a manual repository url to install?
[image]
happy to test!

andrasfuchs, 2023-03-02:
You're right, I can't see version 1.7.1 in KiCad either.
[image]

Somehow [my v1.7.1 update](https://gitlab.com/kicad/addons/metadata/-/merge_requests/170) that I made 2 weeks ago didn't get merged on KiCad's end. I'll contact them.

Until then, I think you should be able to download the newest plugin from here:
https://github.com/freerouting/freerouting/raw/master/integrations/KiCad/kicad-freerouting-1.7.1.zip

Then you can install that .zip file in KiCad's `Plugin and Content Manager` with the `Install from File...` button in the bottom left corner. It's a hassle, I know, sorry about that.

(v1.7.0 didn't check for the path setting in the `plugin.ini`, that was a bug that got fixed in v1.7.1)

kaosbeat, 2023-03-02:
Hi,

installing the zip works fine, however still no go

the jar is is 1.7.0? But if the fix was just for the path, that makes total sense.

```
$ ls -l ~/.var/app/org.kicad.KiCad/data/kicad/7.0/3rdparty/plugins/app_freerouting_kicad-plugin/jar 
total 4656
-rw-r--r-- 1 kaos kaos 4760779 Mär  2 17:27 freerouting-1.7.0.jar
-rw-r--r-- 1 kaos kaos      32 Mär  2 17:27 readme.md

```
one weird thing is that I have another one that was not updated floating around on my system
```

$ ls -l /home/kaos/.local/share/kicad/7.0/3rdparty/plugins/app_freerouting_kicad-plugin/jar
total 4660
-rw-rw-r-- 1 kaos kaos 4760779 Feb 23 12:48 freerouting-1.7.0.jar
-rw-rw-r-- 1 kaos kaos      32 Feb 23 12:48 readme.md

```
I did an uninstall from the GUI and I removed both plugin directories ans reinstalled it. now only the 
`~/.var/app/org.kicad.KiCad/data/kicad/7.0/3rdparty/plugins/app_freerouting_kicad-plugin/`
directory appeared. I edited the path in `  plugin.ini `   but still no go, even after restarting KiCAD

just running 
` java -jar freerouting-1.7.0.jar  
2023-03-02 18:05:13.513 [main] INFO  Freerouting v1.7.0 (build-date: 2023-02-08)
`
seems to work fine (gui is opened)

kaosbeat, 2023-03-02:
Ok I keep looking into it, and I found something weird.
I edited the error message generated in[ plugin.py](https://github.com/freerouting/freerouting/blob/e7ab5f9064b9453103693a84201f1e0cfc30a83d/integrations/KiCad/kicad-freerouting/plugins/plugin.py#L127) when no java is found, directly in the zip file. It does get installed, with the edited message. But it still shows the original message.
If i uninstall the plugin, I cannot run freerouting, the icon is missing. If I install the zip, apparently other code is being run? Can I debug this somehow

andrasfuchs, 2023-03-02:
This is a head-scratcher! 

I'm not familiar enough with the path format on Linux, tbh. Can you run the `java` executable by simply type `/usr/bin/java` to the shell? Or do you need something like a dot (.) or tilde (~) as a first character, like `./usr/bin/java` or `~/usr/bin/java`? 
We have `checkjava.py` and `checkjava.bat` files for Java version testing [here](https://github.com/freerouting/freerouting/tree/master/integrations/KiCad). Could you try to run .py file with the same parameters like you see in the .bat file on your machine?

Having the same `freerouting-1.7.0.jar` with the v1.7.1 plugin is fine.

kaosbeat, 2023-03-02:
for your first question, java can be run simply by typing java, if the /usr/bin folder is in my path environment variable (in a regular shell it is). I can also run it using its full path /usr/bin/java, this should always work, in any shell. The dots or tildes are not needed. ~ is short for home directory, . is short for current directory. in cmbination with a / it is used to prepend a local executable, from preventing to use the global one

```
~$ python -c "import checkjava; checkjava.get_java_version()"
['18.0.2', '18.0.2', '64']
18.0.2

```

dhoepfl, 2023-03-03:
> Ok I keep looking into it, and I found something weird. I edited the error message generated in[ [...]

Maybe python failed to update its cache for some reason. You could try to remove `~/.local/share/kicad/6.0/3rdparty/plugins/app_freerouting_kicad-plugin/__pycache__` and `~/.local/share/kicad/6.0/scripting/plugins/app_freerouting_kicad-plugin/__pycache__` (whichever exist).

> /usr/bin/java is the correct path... If i run the command displayed, copy-pasted into a terminal, freerouter opens and does it magic (the results are however not propagated to my file)

Do you run KiCad in any kind of sandbox/chroot environment?
Does it work if you start KiCad from command line `/usr/bin/kicad`?

kaosbeat, 2023-03-03:
I used a flatpak install. Maybe this is the problem.
I don't know where the binary is, flatpaks are not that transparent to me.

I will try the apt varaiant now (on ubuntu 22.10)

kaosbeat, 2023-03-03:
Ok, flatpak is to blame

```
flatpak uninstall KiCad
sudo apt install kicad

```
solved everything.

andrasfuchs, 2023-03-06:
I (limited) understanding is that because of the Flatpak installation KiCad runs in an isolated container/environment that doesn't have Java installed by default. 

I still need to test this, but this will mean that Freerouting either:
(1) installs Java JRE on the fly when it gets installed via the KiCad Plugin Manager
OR
(2) I need to generate platform-specific self-contained executables of Freerouting that doesn't need the Java JRE

andrasfuchs, 2023-05-02:
Flatpak supports Java JRE installation with [an extension](https://github.com/flathub/org.freedesktop.Sdk.Extension.openjdk17), but KiCad would need to change its installation configuration for this, which I think is unlikely (because it would increase the installation size of KiCad).

(1) I created [a test script](https://github.com/freerouting/freerouting/blob/master/integrations/KiCad/installjava.py) that does the JRE installation on the fly. It could work, but since it needs to download ~50 MBs and needs to extract it which takes some time, we need to have at least a basic user interface showing the process. This interface would be shown if no JRE is detected when we try to launch Freerouting from KiCad. More work needs to be done with this.

(2) I tested [the jpackage tool](https://docs.oracle.com/en/java/javase/20/docs/specs/man/jpackage.html), but it looks like to me that I can't build native executable, I can only create installers that include the Java JRE. It's not useful for us in this case.

andrasfuchs, 2023-05-22:
There is an updated script in the new 1.8.0 release that hopefully solves this problem. It will be available as soon as KiCad merges [my merge request](https://gitlab.com/kicad/addons/metadata/-/merge_requests/217) into their plugin management system.

Zindswini, 2024-12-24:
An alternative method for Flatpak users which does not require the workaround is to install the Flatpak JDK Runtime, and update the freerouting configuration to match, like so:

- Install the Flatpak extension:
    - `flatpak install org.freedesktop.Sdk.Extension.openjdk21`
    - Note: If you installed KiCad system-wide, you must also install the extension as system-wide, or vice verse for a user installation

- And then update the `plugin.ini` to match where Flatpak puts its extensions:
    - Edit `$HOME/.var/app/org.kicad.KiCad/data/kicad/8.0/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.ini`
    - Find the line `path = java`
    - Change it to be `path = /usr/lib/sdk/openjdk21/bin/java`

Maybe this could be mentioned in the popup or added to the KiCad install instructions?