freerouting/freerouting #176: Desire freerouting to use more cores on seriously multi-core system

toddcromiii, 2023-03-01:
Using freerouting 1.7.0, Windows 11 22H2. Using KiCad in general but for now running freerouting standalone using input .dsn file as exported by kicad 7. Tried with and without FR -mt arg and java -xx:activecorecount argument (latest testing: no args of any kind). I expect freerouting to use all cores available, not critical.
--
I have used freerouting for a long time, and it's wonderful (except when it gets stuck and endlessly oscillates, tho in my experience that is likely only when I generate for it a near-impossible routing problem). I've been using it on a i7-2600k system from about 11 years ago.

Problem: Recently I built an AMD 7950X system, and today I tried freerouting doing autorouting for a .dsn file I have. It works, but when I view the activity of the cores in the system (32 cores, using hwinfo or windows resource monitor), about 2 cores are busy at a time, and 30 are near 0 utilization.

Is there a way to get freerouting to use all my cores? Is this a flaw or intrigue of java? Is the routing software problem just too hard to use more cores? Am I misunderstanding? Do any of you actually see FR use many or close to all of the cores present in your system? How?

andrasfuchs, 2023-03-01:
Thanks for the report! 

I had to disable multi-threading by default because it caused many clearance violation issues recently.
You can still change it back by using the `-mt 31`, but I can see that you knew about that already.

You also need to know that there are two part to the routing process: the first is the auto-routing and the second one is route-optimization. The first phase was always single threaded, the second one is implemented in a way that it can be run on multiple threads, that's the `-mt` switch is for.

I understand your issue with this setup, I would love to see freerouting work the fastest way while utilizing all my resources too, but I just don't have the time to rewrite that code at the moment. 

Since this is the case, and it works fine on a single thread, I'll close this issue.

closed 2023-03-01 by andrasfuchs

toddcromiii, 2023-03-01:
That is easily the best response I have gotten recently when inquiring into technical matters; thank you!

Your words explain the behavior I am seeing; in my current test, I posed a difficult problem for FR to solve (one that previous versions have solved, so I know it is possible if I choose the right routing parameters). For this test, it is still working on the initial autorouting phase, so that is why I didn't see the multi-core use when I tried the -mt switch (didn't get to the second, optimization, phase).

In my case, the boards I am producing are small and dense enough that the autorouting phase is often challenging, so I'd love if this got fixed some day. I so appreciate FR (having tried other routing programs) that I plan to look for a way to donate to the FR team.