freerouting/freerouting #177: It's not a problem, it's just praise, it's so cool, please don’t give up this project

hanke-cnc, 2023-03-01:
I just need to put the components, and the rest of the work is given to freeRouting. I use her completely automatic wiring without any problems.
I like this project very much. If there is a donation channel, I will also consider it.

andrasfuchs, 2023-03-01:
@hanke-cnc, I really appreciate it! 
I know that there are many thankful users out there, but it is rare that someone expresses it to me, so thank you! ❤️
A small donation would be nice, you can [check out my sponsors page](https://github.com/sponsors/andrasfuchs) if you wish.
If you have Java or UX/UI skills that would be a great gift too 😉.
Anyway, thank you again, have a great day!

andrasfuchs, 2023-07-20:
Thank you @rozza-m! I really appreciate your words and support, and I welcome you as our first-ever donor! 🥇

cpyne, 2023-09-14:
Agreed.   Best part of designing a board is firing up FreeRouting and clicking "Start Now"

andrasfuchs, 2023-09-16:
Thank you @cpyne for your generosity, it really means a lot!!

closed 2024-03-24 by github-actions[bot]