freerouting/freerouting #179: autoroute no process

EasyEDA2021, 2023-03-24:
To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using? v1.6.2 and v1.7.0 are the same
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? win11 x64
* [ ] What EDA do you use? KiCad, EAGLE, other? EasyEDA Pro
* [ ] What type of file are you working 
[Autorouter_PCB1_2023-3-24.zip](https://github.com/freerouting/freerouting/files/11061759/Autorouter_PCB1_2023-3-24.zip)
with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
* [ ] Did you use any command line arguments?
* [ ] Can you post a screenshot of the problem?
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

we found the attachment fail to auto route, no errors, but it can work at Electra
[Autorouter_PCB1_2023-3-24.zip](https://github.com/freerouting/freerouting/files/11061759/Autorouter_PCB1_2023-3-24.zip)

please help to check

andrasfuchs, 2023-04-24:
I've got the following log when I tried to read the file:

```
2023-04-24 10:50:14.339 [main] INFO  Freerouting v1.7.2-SNAPSHOT (build-date: 2023-04-24)
2023-04-24 10:50:14.355 [main] INFO  Opening '.\tests\Issue179-Autorouter_PCB1_2023-3-24.dsn'...
2023-04-24 10:50:14.933 [main] WARN  Library.read_padstack_scope: the shape of padstack 'p7' is not an area. 
2023-04-24 10:50:14.953 [main] WARN  ShapeSearchTree.calculate_tree_shapes: shape is null
2023-04-24 10:50:15.786 [main] WARN  IntPoint: p_x is out of range
2023-04-24 10:50:15.786 [main] WARN  IntPoint: p_y is out of range
2023-04-24 10:50:15.787 [main] WARN  IntPoint: p_x is out of range
2023-04-24 10:50:15.787 [main] WARN  IntPoint: p_y is out of range
2023-04-24 10:50:15.788 [main] WARN  IntPoint: p_x is out of range
2023-04-24 10:50:15.788 [main] WARN  IntPoint: p_y is out of range
2023-04-24 10:50:15.791 [main] INFO  Opening 'Issue179-Autorouter_PCB1_2023-3-24.rules'...
2023-04-24 10:50:15.791 [main] ERROR File 'Issue179-Autorouter_PCB1_2023-3-24.rules' was not found.
2023-04-24 10:50:15.839 [AWT-EventQueue-0] WARN  GraphicsContext.fill_area: shape not bounded
2023-04-24 10:50:15.948 [AWT-EventQueue-0] WARN  GraphicsContext.fill_area: shape not bounded
```

I changed the the error message a little, because it wasn't too useful before, it didn't tell you which padstack shape is problematic.

Does this help?

EasyEDA2021, 2023-04-26:
> I've got the following log when I tried to read the file:
>
> ```
> 2023-04-24 10:50:14.339 [main] INFO  Freerouting v1.7.2-SNAPSHOT (build-date: 2023-04-24)
> 2023-04-24 10:50:14.355 [main] INFO  Opening [...]

thank you for your help, we will check this issue, thanks

EasyEDA2021, 2023-04-26:
Hi
we check that, it seems the freeeouting doesn't support three points polygon padstack, we will fix this issue at future
thank you for your help

abnormal:
`    (padstack p7
      (shape(polygon 1 0.01 0 -3.5 0 3.5))
    )`

normal:
`    (padstack p7
      (shape(polygon 1 0.01 0 -3.5 0 3.5 2 2))
    )`

closed 2023-05-02 by andrasfuchs