freerouting/freerouting #180: Routing From KiCad 7.0.1 using plugin 1.7.1

mgroz, 2023-03-26:
First off, I really like the new plugin integration in KiCad. Run it and board is routed then pulls everything back into the design. With that being said, you have no way of stepping in on the process if something goes wrong. I would have tried the 1.7.0 plugin but selecting it in plugin manager always installs 1.7.1.

Setup:
Windows 10 (64 bit), KiCad 7.0.1, freerouting plugin 1.7.1

Problem:
I've attached pictures that the names explain what they are. Basically, the 1.7.1 plugin is allowing a trace to be routed on a power plane (GND). I've manually installed the .msi 1.7.0 version and ran the same job manually and it routed the board (correctly) as shown in UsingManual_1.7.0.JPG. As you can see, the J1-2 trace was put on bottom layer B.cu vs In1.cu (GND). 

[Test.zip](https://github.com/freerouting/freerouting/files/11072040/Test.zip)
[SnapShots.zip](https://github.com/freerouting/freerouting/files/11072059/SnapShots.zip)

andrasfuchs, 2023-04-24:
Thank you @mgroz for the detailed report and the files, they are very useful!

So, if I understand it correctly there are 2 or 3 separate issues here:
(1) KiCad installs the latest version of freerouting, no matter which version you select from the Plugin Manager
(2) There is no way to configure freerouting before it runs
(3) There is a problem with the routing itself (that could probably be fixed if we could make freerouting configuration changes before the run?)

mgroz, 2023-04-27:
Andras,Yes to all points but It would be nice to configure all settings in KiCad before launching Freerouting, however, tweaks maybe necessary if routing cannot complete. I think a check for this incomplete routing before exiting back to KiCad would allow the choice to handle it in Freerouting or just save the routing and exit back with current route results. Once back in KiCad, the user could change parameters and re-run Freerouting to complete or manually finishing routing in KiCad. Whats nice to this approach is all routing parameters remains in KiCad. I hope my explanation made sense. Sent from Mikes iPad

andrasfuchs, 2023-05-04:
I have just check, and (1) is fixed in KiCad 7.0.2.

For (2) I added a new option to make configuration changes before the auto-routing starts.

[image]

You can wait for the next release or test it with the latest snapshot build.

mgroz, 2023-05-04:
Maybe I was not clear in my previous post. The idea of router configuration can get confusing since these settings can come from 2 places, KiCad or Freerouting. I’m just trying to ensure the import into Freerouting takes the currently configured settings already established in KiCad.

andrasfuchs, 2023-05-04:
Ah, alright, I did misunderstand it then. 

Could you give some guidance how to test this? What settings from KiCad are not transferred to Freerouting, and how can I set them in KiCad/pcbnew? Are they saved in the DSN file?

mgroz, 2023-05-07:
It appears that the trace width and clearance settings are transferring over to Freerouting (via netclass). I've found a workaround for obtaining previous versions of the plugin. Just select it then download it first. From the plugin box, use install from file.

Another problem has come up with KiCad 7.0.2. When using a properly defined mounting hole (6-32 size) that has a 150 mil **NPTH** hole with a 320 mil keepout area, a large shape appears in Freerouting (see pic). If I replace the NPTH with a 150 mil round **Edge.Cut** hole, it works fine. This pad is also ignoring the board Edge.Cuts polygon as well. This prevents any further routing. 

I've attached 2 DSN files: Test_NPTH.dsn (uses NPTH hole) and Test_EdgeCuts (use cut-out hole).dsn. 

Unfortunately, KiCad has changed their NetClass Interface, see OLD and NEW. For manual routing, you can define track size and via clearances on the fly but they do not stick with the net. The new interface requires you to manually type the entire net out when defining the class. Not sure why they couldn't do both...

Mounting Hole:
[image]
KeepOut Rules:
[image]
Large Pad Coming Accross In DSN File:
[image]
Test+EdgeCuts.DSN:
[image]
NetClass OLD:
[image]
NetClass NEW:
[image]

[Test.zip](https://github.com/freerouting/freerouting/files/11413760/Test.zip)

andrasfuchs, 2023-05-08:
The large pad issue is discussed here: https://gitlab.com/kicad/code/kicad/-/issues/14656

I'm still not sure about the other issue you have, sorry.

> It appears that the trace width and clearance settings are transferring over to Freerouting (via netclass).

That's the expected behavior, right?

> I've found a workaround for obtaining previous versions of the plugin. Just select it then download it first. From the plugin box, use install from file.

Which version?

lynchaj, 2023-05-08:
HiI have since fixed the padless hole problem by replacing H3 and H4 with holes with pads so that problem is solved. 
The issue is when I load a .DSN file that's been exported from KiCAD based on a previously imported .SES file, I get hundreds of clearance violations.  This happens with all the versions of FreeRouting I use ranging from 1.2.43 Extra to 1.7.0 to the latest snapshot.
Steps to reproduce the problem:1. start with PCB in KiCAD with parts but no traces.  If PCB has traces, you can select all traces & vias and delete them2. export .DSN from KiCAD3. load .DSN in FreeRouting, perform trace routing and optimization4. save .SES file from FreeRouting5. import .SES file into KiCAD (adds traces to PCB)6. export .DSN from KiCAD (parts plus traces)7. load .DSN in FreeRouting, press "violations" button to see clearance violations. Should be hundreds of them.  Also "incompletes" button will show several incompletes (10-12 usually).  If I use the latest snapshot, I get about 400 violations.  If I use 1.2.43 Extra I get about 1600 on current iteration of Z80 processor
You should be able to replicate my findings with the Z80 processor zip file I uploaded previously
Thanks, Andrew Lynch

andrasfuchs, 2023-05-09:
Yes, and it looks like that we can expect a fix for this in KiCad 7.0.3 👍 Meanwhile you can use 7.0.1.

[image]

closed 2023-05-22 by andrasfuchs