freerouting/freerouting #181: I downloaded the latest - 20230301

benirieth, 2023-03-30:
To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
* [ ] What EDA do you use? KiCad, EAGLE, other?
* [ ] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
* [ ] Did you use any command line arguments?
* [ ] Can you post a screenshot of the problem?
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

benirieth, 2023-03-30:
Im using the latest version of kiCAD and freerouting as well. 20230301
Windows 10 x64
its a dsn file
i didnt use any command line arguements
I tried to open an other file. That could handle it. It contain a bit more parts 155x95 mm the PCB size, approx 100 components

andrasfuchs, 2023-04-24:
Thanks @benirieth for the report. 

I can't help you without the KiCad project files or the DSN file. Could you provide them you so that I can look into this?

I've just made a few fixes to our snapshot build, so alternatively you could try to open the file with that.

closed 2023-05-22 by andrasfuchs