freerouting/freerouting #185: Following KiCad upgrade to 7.0.2, no more autorouting...

chf-6309, 2023-04-17:
To make our work easier, please provide as many details as you can:
* [ x ] What version of freerouting are you using? Freerouting v1.7.1
* [ x ] On what platform did you run it? Windows,  64-bit, latess patchs and updates, tested on two different LapTop
* [ x ] What EDA do you use? KiCad, upgraded from 7.0.1 to 7.0.2
* [ x ] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file? Uce a microprocessor pcb, that autoroute well on KiCad 7.0.1 and earlier versions
* [ x ] Did you use any command line arguments? no, use the KiCad plugin...
* [ . ] Can you post a screenshot of the problem? Not now...
* [ . ] Can you include the latest, related log entries from the .log file? where are the log locateds?
* [ . ] Please list the detailed steps to reproduce the problem: open an PCB project, then clic FreeRoute, the dsn files are generated, FreeRoute open (JRE 17.0.10 installed, as recommended), the a "new" blue windows appear, a few seconds , with the spaghetti connections, a few seconds reflexion, tyhe no autoroute and come back to the unrouted PCB
* [ X ] What would you expect freerouting to do in this situation? route the PCB connections
* [ X ] Is fixing this issue critical for your work? of course!

One more suggestion(s)
* may a delay before the autoroute start automatically, so, no "adjustments" could be made before the route start
* during the routing compute, I would prefer a black background, not a dark grey: is there an option to memoryse my "global" settings?
Thanks for your work.

DeflateAwning, 2023-04-22:
The autorouter is crashing in KiCAD 7.0.2 (which I think is what this user is reporting). When you launch the freerouter, it loads, then closes itself 3-10 seconds later. You can briefly see the unrouted ratsnest, and then it closes.

andrasfuchs, 2023-05-02:
Yes, I can confirm that there is a fatal problem with the DSN exporter in KiCad in version 7.0.2.

This is the diff of the exported DSN file in one of the test projects (red is done by KiCad 6, green is by KiCad 7):
[image]

As you can see the coordinates are written incorrectly, there is an extra 500 before the correct 650 value.

chf-6309, 2023-05-02:
Hello Andras, 

thank you very much for your successful investigations! 
So, it seems that the issues seats on some DSN export garbage that offset the real values. 

May I have to send a bug information to KiCad, or maybe You did it already? 

Thanks for your work and commitment to this auto-router: it make the microprocessors PCB easy to hand correct/optimization, after a successful FreeRouting. 

Kind Regards. 
-=chf=- 

----- Mail original -----

image

DeflateAwning, 2023-05-03:
Even though it seems like the main issue is upstream in KiCad v7, it would be way better if the autorouter didn't appear to crash when it encountered these sorts of errors. Perhaps there's a good follow-up action to address the secondary issue within this library as well

andrasfuchs, 2023-05-03:
Yes, I agree, we need much better error reporting. That is one of the three planned improvement for the next release.

heikkiw, 2023-05-04:
Hi, I have also the same problem, and I can verify that this is actually Kicad 7.0.2 issue.
It is related to using Kicad keepout zones. If you have a PCB design without keepout zones Freerouter work as expected. I verified this by manually deleting "keepout circle" lines from the DSN-file. So workaround to this issue is not use keepout zones or delete these from DSN-file. But ofcourse you must manually reroute the tracks that are in the keepout areas ! Hope we can get fix to this problem asap.

andrasfuchs, 2023-05-04:
@heikkiw Thank you for the confirmation!

Meanwhile I added a new DSN file reader log window that is displayed just after Freerouting tries to load the exported file, so we can at least see what is the reason for not loading/not autorouting the board.

[image]

closed 2023-05-22 by andrasfuchs