freerouting/freerouting #189: Freerouting exception on Ubuntu 23.04

davidsrsb, 2023-05-04:
After a report of an exception on 23.04, I tried from the command line and get

java -jar freerouting-1.7.0.jar 
Exception in thread "main" java.lang.UnsatisfiedLinkError: Can't load library: /usr/lib/jvm/java-17-openjdk-amd64/lib/libawt_xawt.so
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2393)
	at java.base/java.lang.Runtime.load0(Runtime.java:755)
	at java.base/java.lang.System.load(System.java:1953)
	at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
	at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232).......

This may be related
https://stackoverflow.com/questions/72049332/java-lang-unsatisfiedlinkerror-opt-jdk17-lib-libawt-xawt-so-libxrender-so-1

davidsrsb, 2023-05-04:
Installing the default-jre package seems to fix this.
Ubuntu 23.04 iat upgrade installs the headless jre

andrasfuchs, 2023-05-08:
Thank you @davidsrsb for the report, that's good to know!