freerouting/freerouting #191: Clearance violations for certain boards

andrasfuchs, 2023-05-04:
There are still cases where Freerouting produces clearance violations. They should not happen.

If you have a KiCad project file that typically gets clearance violations after auto-routing, please upload it here.

benwillcox, 2023-05-07:
I've just found this issue after upgrading from 1.4.x (can't remember the exact version I'm sorry) having been using this version on various KiCAD projects with no issues. After installing 1.6.5 I found multiple clearance issues after autorouting a project that worked fine in the old version.
I have rolled back to 1.4.5 and it routes the same project with no clearance problems :-)

andrasfuchs, 2023-05-08:
Thank you @benwillcox for the detailed report, it is great input! I'm glad your case works with 1.7.0!

I'll keep this issue open nonetheless until I tested it with multiple boards.

andrasfuchs, 2023-05-08:
I started to investigate the issue, and I can confirm that auto-routing causes violations with the Z80 board.

I run it a few times, and so far it looks like that Freerouting has problems with the two hole clearances, H3 and H4, it violates the 0.25mm clearance for both.

[image]

It's strange, because I can see the copper to hole clearance definition of 0.25mm here:
[image]

But when KiCad exports the board, there is no mention of this rule, at least I can't seem to find it.
Is it normal to not export this rule to the DSN or is it a bug in KiCad?

lynchaj, 2023-05-08:
Hi
Yes, those are known issues and not the clearance violations I am concerned about.  Both H3 and H4 are padless holes for routing wires between front and back of PCB if needed during debugging. I have since replaced both padless holes with holes with pads and since it then appears on the copper layer, FreeRouting is able to handle them properly.  The silkscreen clipping is by design because the IDC connectors descend lower than the edge of the PCB.  I handle those as exclusions to the DRC run.

The clearance violations I am concerned about are if you load the .SES file into KiCAD and then save it again as a .DSN file.  Load the resulting .DSN file in FreeRouting and you'll see about 1500 clearance violations.  At least that's what I see when I try it.

Thanks, Andrew Lynch

andrasfuchs, 2023-05-09:
Hi @lynchaj ,

Thank you for the detailed steps, I could reproduce the problem now, and started the investigation. Something isn't right, I'm just not sure yet where and if KiCad or Freerouting causes the problem.

Meanwhile I extended the clearance violation details in Freerouting a little, so we can see how serious the violation is:
[image]

It looks like a rounding problem to me, but I also noticed something curious. If I set a specific clearance value in KiCad, it saves just a slightly higher value into the DSN. I don't think it's intentional, but who knows.

Settings in KiCad:
[image]
Values saved in DSN:
[image]

I'll continue the testing and report my findings here.

andrasfuchs, 2023-05-09:
I use 64-bit versions of both KiCad and Freerouting.

I suspect that both a KiCad DSN export problem and our collision detection algorithm are to blame. 
It's still a little murky for me, so I'll need to dig deeper for this.

The good news is that it looks like that the violations are tiny, so although the violation list is annoying, they should not cause any issues through fabrication.

lynchaj, 2023-05-09:
HiPersonally, I think this issue has been around for a long time. I've been using KiCAD and FreeRouting since 2006 and this has been a recurring problem as long as I can remember.
The main problem this causes is it prevents an iterative design cycle to improving the board in small increments. Essentially, the transfer from KiCAD to FreeRouting and back to KiCAD is a one-time event.  Attempting to iterate will introduce clearance violations and subsequent cycles makes it worse.
KiCAD DSN -> FreeRouting -> KiCAD SES -> the end
What I would like to be able to do is repeat in a cycle as needed to make adjustments without starting over from scratch.  Being able to preserve the >90% trace routing solution but still able to make minor PCB updates and even manually adjust traces in KiCAD and re-enter the FreeRouting optimization.
Hopefully, that is possible or something close.  Your help is greatly appreciated!

Thanks, Andrew Lynch
Message ID: ***@***.***>

andrasfuchs, 2023-05-09:
I agree that it would be desirable, round-trips between KiCad and Freerouting should be a non-issue. Your use-case is a totally valid one.

As it turns out, we had a good suspicion earlier, there is a backed-in safety margin setting in KiCads DSN exporter, that might be the reason why it's not working properly. See the discussion here: https://gitlab.com/kicad/code/kicad/-/issues/14713 .

If they agree to remove that feature in their next release and I implement our own safeguards in Freerouting's next release then I don't see why it would be not possible to round-trips. I can't promise anything, because anything can come up during the process that I don't know about, but it's worth a try.

I'll keep you posted here.

lynchaj, 2023-05-09:
Hi Andras,
Thank you so much!  I think this would be a great fix to KiCAD and FreeRouting.  Even if it takes a while, at least it is acknowledged as something that needs fixing.  I think this issue has been around for quite a while (2009?).

Thank you!  Much appreciated!  Andrew Lynch

andrasfuchs, 2023-05-15:
Alright, so I spent a good amount of time on testing and figuring out what the root of the issue can be.

My first suspicions were the SES and DSN file writers and DSN file readers, I thought that they do something funny with rounding the margins or coordinates. Although I cleaned up the DSN and SES writer routines I don't think they were the issue. There were some rounding hacks built-in, but they had no impact on the clearance violations.

The differences between the clearance calculation methods of KiCad and Freerouting are most likely to blame. The practical implications of this are so tiny that I decided to add an internal safety margin to the router, so it will not max out the allowed clearances that are defined in the net classes. This extra margin is 0.0016 mm, so I don't think that it will ever cause any problems for fabrication.

Anyway, after doing some more tests with KiCad 6 and Z80 prototype, clearances are still good even after a dozen roundtrips between KiCad and Freerouting.

Could you test [the snapshot build](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT) and give some feedback?

lynchaj, 2023-05-15:
Hi,
Will do and report back later, hopefully today

Thanks!  Much appreciated, Andrew Lynch

lynchaj, 2023-05-16:
Hi
I am afraid the problem is still there.  I went through the cycle of saving the .SES file from FreeRouting and then importing into KiCAD and then exporting the .DSN file.  In KiCAD, the board passes DRC without any problems other than the known 12 clipping issues from the IDC connectors.  When I loaded the .DSN file into the latest snapshot FreeRouting, I had 2062 clearance violations (basically the whole board was covered in clearance violations) and 8 incompletes.

I zoomed in on the clearance violations and noticed some occur in the middle of traces.  I am wondering if there is some issue in KiCAD where trace segments join if there is some overlap which appears as clearance violation FreeRouting?  I also zoomed in on the 8 incompletes and they don't make much sense either.  The traces are present and seem to go from the starting to finish points but FreeRouting says they are not connected.  ???

Sorry, I wish I had better news.  Thanks, Andrew Lynch

lynchaj, 2023-05-16:
[processor-Z80-test.zip](https://github.com/freerouting/freerouting/files/11486997/processor-Z80-test.zip)

This is my test .DSN file

andrasfuchs, 2023-05-16:
Alright, no worries, I'll keep looking. Thank you for the details!

I'll try to spare some time for this later this week.

lynchaj, 2023-05-21:
Hi
One additional thing I've noticed with this board is that if it runs in the FreeRouting trace optimizer for an extended period of time like 1 week or more when I try to save the board I get "Java.Lang.StackOverflowError" message.  Prior to the one week point, it saves fine.

Once a day or so, I interrupt the trace route optimization to export an .SES file, save as a .DSN file, and "save" the board.  The "save" option writes a binary file .BIN however I can only do the save function for the first few days.  After about a week I get the Java Stack overflow error.

This is the same .DSN file as I've posted earlier in the week.

Thanks, Andrew Lynch

andrasfuchs, 2023-05-22:
Thank you Andrew! 

It's a useful feedback, but I don't think I will be able to test this with such a long running times. 

Since this seems to be independent from this issue (clearance violations), please create a new, separate issue if it still happens with the newest version.

lynchaj, 2023-09-28:
Hi
Was there ever a root cause identified for the clearance violations seen in export of .DSN, FreeRouting, export of .SES, and import of .SES into KiCAD, and re-export of .DSN file to FreeRouting?

Thanks, Andrew Lynch

lynchaj, 2023-11-28:
Hi
Please keep researching this issue.  FreeRouting should be able to import boards exported from KiCAD with DRC compliant tracing without generating hundreds of clearance violations.  Something is dreadfully wrong in FreeRouting that it is not able to import legitimate, error-free boards KiCAD exports as DSN files.  Thank you, Andrew Lynch

PS,  this is not a "once in a while" problem.  Practically every board I make in KiCAD cannot be reimported into FreeRouting once the traces exist without hundreds of clearance violations.  This makes no sense to me.

lynchaj, 2024-03-09:
Hi, still an issue.  Hopefully someone can find and fix this issue.  Thanks!

closed 2024-05-16 by github-actions[bot]