freerouting/freerouting #194: freerouting cannot load gnome-shell extension

zapphyre, 2023-05-14:
dumbfounded i stare at this error:
```
Program failure:
---
command:
java -jar /home/tepo/.local/share/kicad/7.0/3rdparty/plugins/app_freerouting_kicad-plugin/jar/freerouting-1.7.0.jar -dl -de /home/tepo/Documents/incubo-kicad/freerouting.dsn -do /home/tepo/Documents/incubo-kicad/freerouting.ses
---
exit code: 1
--- stdout ---
b''
--- stderr  ---
b'\n(java:12302): Gtk-WARNING **: 01:16:48.415: Theme parsing error: gtk.css:1:106: Failed to import: Error opening file /home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet.css: No such file or directory\n\n(java:12302): Gtk-WARNING **: 01:16:48.415: Theme parsing error: gtk.css:2:112: Failed to import: Error opening file /home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet-tiled.css: No such file or directory\n'
```
it appears when I'm starting `freerouting` (which is great otherwise)  from KiCad 7.0.1
it is bizarre to me because:

- it has been working great a few days ago. since then I did nothing to my system
- I don't have such plugin installed and never had
- it started to do the same thing on the other computer with OSX version

it's really bizarre to me, which may point to me being issue, but environment is quite diversified (2 pcs, 2 oses) and it was working very recently. 

OSX monterey & Archlinux

any suggestions?  
thanks!

andrasfuchs, 2023-05-15:
Strange indeed! Can you run other Java apps or do you have the same issue with them?

Did you use Flatpak to install KiCad?

**I asked Bing this was its answer:**
It seems like you are getting a `Gtk-WARNING` due to a `Theme parsing error`. This error occurs when the system is unable to find the specified file or directory. 

One solution could be to check if the file `~/.config/gtk-3.0/gtk.css` contains `@import url (filename)` statements, referring to non-existent files in `/extensions/no-title-bar`. If it does, you can remove the erroneous import statements from the file `gtk.css` ³. 

If that doesn't work, you can try setting your GTK theme again in System Settings > Application Style ¹. Let me know if this helps! 😊

**And I also asked ChatGPT:**
The error message you provided suggests that there is a problem with importing certain CSS files related to the GTK theme. Specifically, it seems that the files `/home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet.css` and `/home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet-tiled.css` cannot be found.

To resolve this issue, you can try the following steps:

1. Check if the files mentioned in the error message actually exist in the specified location. You can use a file manager or run the following commands in the terminal:

```bash
ls /home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet.css
ls /home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet-tiled.css
```

If the files are missing, you may need to reinstall the corresponding extension.

2. If the files are present, it's possible that there is a permissions issue preventing the application from accessing them. Ensure that the files have appropriate read permissions for your user. You can use the following command to add read permissions if needed:

```bash
chmod +r /home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet.css
chmod +r /home/tepo/.local/share/gnome-shell/extensions/no-title-bar@jonaspoehler.de/stylesheet-tiled.css
```

3. If you recently updated or changed the GTK theme or related extensions, it's possible that the references to those CSS files haven't been updated. In that case, you may need to update the configuration or settings of the application or theme in question.

If none of the above steps resolve the issue, it might be helpful to provide more details about the application or system you are using and any recent changes you made that could have triggered this error.

zapphyre, 2023-05-15:
it's just stupid error cases drag in attention of state-clerk like people.

closed 2023-05-23 by andrasfuchs