freerouting/freerouting #196: HOWTO use CodeSee Maps?

abelom, 2023-05-15:
just attempted installing CodeSee Maps did it go right? Where does one see any results/features/consequences?

andrasfuchs, 2023-05-15:
It looks good, but I need to learn more about it to make it useful. Do you have any experience with it?

[image]

I hope it helps with the refactoring effort that would be needed for [the headless CLI version](https://github.com/freerouting/freerouting/discussions/195).

andrasfuchs, 2023-06-05:
CodeSee didn't work out in the end, Shanea Leven, the founder couldn't resolve our technical issue.

I don't seem to have the right to remove its access to the repo, @abelom could you take a look?
https://github.com/apps/codesee-maps/installations/37536226

abelom, 2023-06-06:
``You're all set. A job has been queued to uninstall CodeSee Maps  ``

closed 2023-06-06 by abelom