freerouting/freerouting #207: Java Error (non-critical)

neltnerb, 2023-06-10:
To make our work easier, please provide as many details as you can:
* [1.7.2-SNAPSHOT (build date: 2023-05-15), Adoptium JRE 17 (also tried with JRE 20 with same message)] What version of freerouting are you using?
* [Linux Ubuntu 22.04 64-bit] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
* [KiCad 7.0] What EDA do you use? KiCad, EAGLE, other?
* [.kicad_pcb] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
* [not intentionally, not sure what kicad invokes] Did you use any command line arguments?
* [
[image]
] Can you post a screenshot of the problem?
* [I am not sure which log entries would be related, do you need me to find a log from the JRE?] Can you include the latest, related log entries from the .log file?
* [Open PCB in KiCAD, open the freerouting addin and this message appears while loading the program] Please list the detailed steps to reproduce the problem
* [Not show that message (it seems to do nothing to the functionality)] What would you expect freerouting to do in this situation?
* [No, the tool still works fine, the error is just weird.] Is fixing this issue critical for your work?

rei-vilo, 2023-06-18:
Same issue here, with 

```
$ java --version
openjdk 17.0.7 2023-04-18
OpenJDK Runtime Environment (build 17.0.7+7-Debian-1deb12u1)
OpenJDK 64-Bit Server VM (build 17.0.7+7-Debian-1deb12u1, mixed mode, sharing)

$ uname -a
Linux 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux
```

andrasfuchs, 2023-10-04:
It should be fixed now, you can try the lastest snapshot build:
https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

closed 2023-10-04 by andrasfuchs

rei-vilo, 2023-10-04:
Tested successfully, thank you!