freerouting/freerouting #208: Freerouting / Kicad 7 not doing anything

atlaste, 2023-06-13:
To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?

Freerouting 1.8.0

* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?

Windows, 64-bit.
openjdk 17.0.7 2023-04-18
OpenJDK Runtime Environment Temurin-17.0.7+7 (build 17.0.7+7)
OpenJDK 64-Bit Server VM Temurin-17.0.7+7 (build 17.0.7+7, mixed mode, sharing)

* [ ] What EDA do you use? KiCad, EAGLE, other?

KiCAD 7.0.2

* [ ] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?

Renamed as .txt and uploaded. I've tried to make it as simple as possible.
[freerouting.dsn.txt](https://github.com/freerouting/freerouting/files/11731945/freerouting.dsn.txt)

* [ ] Did you use any command line arguments?

Nope. I looked for logging options in "-h", but there don't appear to be any? The command line doesn't show anything. I'm happy to provide a log file if I can find it...

* [ ] Can you post a screenshot of the problem?

I'll describe the behavior. If I run freerouting, it quickly moves all the 'to route' to 'failed' and once the counter hits 0, it crashes. Nothing routed, nothing happened, no error.

Tried it on a laptop and my desktop; both same behavior.

* [ ] Can you include the latest, related log entries from the .log file?

Yea if I could find it...

* [ ] Please list the detailed steps to reproduce the problem

See above

* [ ] Is fixing this issue critical for your work?

Mwah, I'm not looking forward to manually routing a gazillion connections...

andrasfuchs, 2023-06-13:
I took a quick look and it seems like a KiCad issue. Version 7.0.2 had a bug that produced invalid DSN files, those big blue ovals are not normal. 

[image]

Could you try [the newest version](https://www.kicad.org/download/)? 
Someone here already confirmed that it fixed clearance issue.

atlaste, 2023-06-13:
I'll give it a shot and get back to you. Thanks for the response.

atlaste, 2023-06-13:
Thanks, that seemed to have fixed the issue at hand. 
I also noticed Freerouting has improved significantly since the last time I've tested it. Great job, I like it!