freerouting/freerouting #214: Freerouting dies with error and warning messages when run from KiCad plugin

ned-pcs, 2023-07-11:
I'm getting an error upon trying to run Freerouting from KiCad. It opens a Java window with two warnings, then an error dialog.
I have also tried running Freerouting from the command line with the same .DSN file with the same results.
I have attached the .DSN file.

* What version of freerouting are you using? 1.8.0
* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? Linux 64-bit x86
* Java details:
```
openjdk 17.0.7 2023-04-18
OpenJDK Runtime Environment Temurin-17.0.7+7 (build 17.0.7+7)
OpenJDK 64-Bit Server VM Temurin-17.0.7+7 (build 17.0.7+7, mixed mode, sharing)
```
* What EDA do you use? KiCad, EAGLE, other? KiCad 7.0.5
* What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
 [freerouting_bug.zip](https://github.com/freerouting/freerouting/files/12009548/freerouting_bug.zip)
* Did you use any command line arguments? KiCad did:
[image]
* Can you post a screenshot of the problem? Yes.
```
WARNING Package.read_rotation: closing bracket expected
WARNING Package.read_pin_info: String or Integer expected
```
* Can you include the latest, related log entries from the .log file? Yes, here:
[freerouting.log](https://github.com/freerouting/freerouting/files/12009537/freerouting.log)
* Please list the detailed steps to reproduce the problem
* What would you expect freerouting to do in this situation?
* Is fixing this issue critical for your work? No, I can manually route the remainder of the board.

andrasfuchs, 2023-09-28:
Sorry for the late reply @ned-pcs.

I had some time to check out your issue today. I improved freerouting's error logging, so that we can find the component it has a problem with:

[image]

I opened the DSN file and looked for that name and I found this:
[image]

The problem here is that the number followed by the `rotate` keyword is `2e-08` and we can't parse it that way at the moment.

According to the Specctra specification numbers in a DSN file must be C-language floating-point numbers.
[image]

At this point I'm not sure if we should implement the parsing for this format, or ask the KiCad developers to save the rotation information in the same way they do with all the other numerical values.

As for a workaround tip for you: this rotation value seems to be so tiny that it's likely that it got there by accident, so you can check that particular component in KiCad, and remove its rotation value.

ned-pcs, 2023-09-28:
Thanks for diagnosing this. I will fix the rotation values. I have no idea how this happened.

andrasfuchs, 2023-10-03:
Alright, I made a new parser for the rotation value, so the format like `2e-08` should work without any issues in the next release.

closed 2023-10-04 by andrasfuchs