freerouting/freerouting #218: ClassCastException: class javax.swing.JSeperator cannot be cast to class javax.swing.JPanel

makuser, 2023-08-17:
To make our work easier, please provide as many details as you can:
* [x] What version of freerouting are you using?
1.8.0
* [x] On what platform did you run it?
Linux Ubuntu 22.04.3 LTS
* [x] What EDA do you use?
KiCad 6.0.2+dfsg-1
* [x] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
Autogenerated files by KiCad plugin
* [x] Did you use any command line arguments?
`java -jar /home/username/.local/share/kicad/6.0/3rdparty/plugins/app_freerouting_kicad-plugin/jar/freerouting-1.8.0.jar -dl -de /project/freerouting.dsn -do /project/freerouting.ses`
* [x] Can you post a screenshot of the problem?
[image]
* [ ] Can you include the latest, related log entries from the .log file?
Where would I find those?
* [x] Please list the detailed steps to reproduce the problem
- open pcbnew (empty/unopened file is fine) with installed freerouting plugin
- click on freerouting button to generated DSN files and open freerouting automatically
- freerouting will show an error right away
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

daniel-theia, 2023-08-24:
The issue is in `src/main/java/app/freerouting/gui/MainApplication.java:370`:

```java
JButton startButton = (JButton)((JPanel)optionPane.getComponents()[1]).getComponents()[0];
```

There appears to be a JSeperator where a JButton is expected.

Have created the following PR that fixes this issue: https://github.com/freerouting/freerouting/pull/220

geenweb, 2023-09-04:
I'm having the exact same issue.
Freerouting 1.8.0
KiCad Version 7.0.7
Ubuntu 20.04.1
Linux Mint 20

This is fatal, so I'm hoping for a new release soon so I can try Freerouting.

Thanks.

daniel-theia, 2023-09-21:
@makuser @geenweb If possible, please try building `master` to test the merged PR: https://github.com/freerouting/freerouting/pull/220

akhodeir, 2023-09-23:
I have the same issue.
were solutions above fixed the issue ?

daniel-theia, 2023-09-25:
> I have the same issue. were solutions above fixed the issue ?

@akhodeir Yes, this should fix this issue.

Sfinx, 2023-10-01:
Trying to build the latest git (c784202298c66) with this fix (242caac) included but see weird thing - src/main/java/app/freerouting/gui/MainApplication.java file in master branch do not have this fix despite it is labeled as merged. WTF ?

andrasfuchs, 2023-10-04:
It should be fixed now, you can try the lastest snapshot build:
https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

closed 2023-10-04 by andrasfuchs

Sfinx, 2023-10-04:
Yep, it fixed ! Thanks !

andrasfuchs, 2023-10-17:
The plan is to have the new release in 2 weeks, followed by the KiCad plugin update soon after.