freerouting/freerouting #219: java.lang.NullPointerException: Cannot read field "net_no_arr" because "curr_trace" is null

clydeshaffer, 2023-08-20:
To make our work easier, please provide as many details as you can:
* [x] What version of freerouting are you using?
v1.8.0
* [x] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Windows 11 64-bit
* [x] What EDA do you use? KiCad, EAGLE, other?
KiCad
* [x] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
DSN file exported from KiCad
[LogicBoard_smt.zip](https://github.com/freerouting/freerouting/files/12388529/LogicBoard_smt.zip)
* [x] Did you use any command line arguments?
No
* [x] Can you post a screenshot of the problem?
[image]
* [x] Can you include the latest, related log entries from the .log file?

[freerouting.log](https://github.com/freerouting/freerouting/files/12388530/freerouting.log)

`2023-08-20 11:37:20.338 [Thread-0] ERROR Cannot read field "net_no_arr" because "curr_trace" is null
java.lang.NullPointerException: Cannot read field "net_no_arr" because "curr_trace" is null
	at app.freerouting.board.PullTightAlgo.smoothen_end_corners_at_trace_1(PullTightAlgo.java:446) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.board.PullTightAlgo.opt_changed_area(PullTightAlgo.java:158) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.board.RoutingBoard.opt_changed_area(RoutingBoard.java:227) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.board.RoutingBoard.opt_changed_area(RoutingBoard.java:184) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.board.RoutingBoard.autoroute(RoutingBoard.java:1014) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.interactive.SelectedItemState.autoroute(SelectedItemState.java:353) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.interactive.InteractiveActionThread$AutorouteThread.thread_action(InteractiveActionThread.java:83) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.interactive.InteractiveActionThread.run(InteractiveActionThread.java:49) ~[freerouting-executable.jar:unspecified]
`
* [x] Please list the detailed steps to reproduce the problem
Used rules->net classes to select the kicad_default net class, and then began autorouting that netclass
* [x] What would you expect freerouting to do in this situation?
Not cause NullPointerExceptions in the middle of routing
* [x] Is fixing this issue critical for your work?
Yes

clydeshaffer, 2023-08-20:
Same issue also happens using the 1.7 snapshot that downloads via the KiCad plugin manager.

clydeshaffer, 2023-08-20:
Also appears that this doesn't happen if I don't select the kicad_default net, but I need to exclude the Power net so I can do it manually.

andrasfuchs, 2023-10-04:
Thanks @clydeshaffer for the report.

I'm trying to reproduce the problem, but I can't seem to reach the point where I get that exception. How exactly do you select the `kicad_default` net class and what buttons to you click on after that?

Also, there is a function in freerouting on the same Rules / Net Classes window, where you can check the "ignored by autorouter" column for your Power net class. Would that do the trick?

[image]

clydeshaffer, 2023-10-04:
Unfortunately I don't remember why "ignored by autorouter" wasn't working for me. I think it was trying to route my power nets despite the setting, and at the time I assumed the option must have been for some other unknown purpose.

I was using the "Select" button in this window, which would highlight all the nets I wanted to route. Once the nets were highlighted, the options at the top of the main window would change and include an "Autoroute" button with the tooltip "to autoroute the selected items (a)". It would then work for a bit but consistently fail at seemingly the same place. "to route: 217" was consistently displayed on the status bar when the error was encountered.

[image]

I ended up getting around the issue with this code change:
```
diff --git a/src/main/java/app/freerouting/board/PullTightAlgo.java b/src/main/java/app/freerouting/board/PullTightAlgo.java
index 9a0c120..4ce2cb5 100644
--- a/src/main/java/app/freerouting/board/PullTightAlgo.java
+++ b/src/main/java/app/freerouting/board/PullTightAlgo.java
@@ -429,34 +429,37 @@ public abstract class PullTightAlgo {
       connection_to_trace_improved = false;
       Polyline adjusted_polyline = smoothen_end_corners_at_trace_2(curr_trace);
       if (adjusted_polyline != null) {
-        result = true;
-        connection_to_trace_improved = true;
         int trace_layer = curr_trace.get_layer();
         int curr_cl_class = curr_trace.clearance_class_no();
         FixedState curr_fixed_state = curr_trace.get_fixed_state();
         board.remove_item(curr_trace);
-        curr_trace =
-            board.insert_trace_without_cleaning(
+        PolylineTrace adj_ins_trace = board.insert_trace_without_cleaning(
                 adjusted_polyline,
                 trace_layer,
                 curr_half_width,
                 curr_trace.net_no_arr,
                 curr_cl_class,
                 curr_fixed_state);
-        for (int curr_net_no : curr_trace.net_no_arr) {
-          board.split_traces(adjusted_polyline.first_corner(), trace_layer, curr_net_no);
-          board.split_traces(adjusted_polyline.last_corner(), trace_layer, curr_net_no);
+        if(adj_ins_trace != null) {
+          result = true;
+          connection_to_trace_improved = true;
+          board.remove_item(curr_trace);
+          curr_trace = adj_ins_trace;
+          for (int curr_net_no : curr_trace.net_no_arr) {
+            board.split_traces(adjusted_polyline.first_corner(), trace_layer, curr_net_no);
+            board.split_traces(adjusted_polyline.last_corner(), trace_layer, curr_net_no);
 
-          try {
-            board.normalize_traces(curr_net_no);
-          } catch (Exception e) {
-            FRLogger.error(
-                "The normalization of net '" + board.rules.nets.get(curr_net_no).name + "' failed.",
-                e);
-          }
+            try {
+              board.normalize_traces(curr_net_no);
+            } catch (Exception e) {
+              FRLogger.error(
+                  "The normalization of net '" + board.rules.nets.get(curr_net_no).name + "' failed.",
+                  e);
+            }
 
-          if (split_traces_at_keep_point()) {
-            return true;
+            if (split_traces_at_keep_point()) {
+              return true;
+            }
           }
         }
       }
```