freerouting/freerouting #221: `-is random` Ignores Violations

daniel-theia, 2023-08-30:
To make our work easier, please provide as many details as you can:

* [x] What version of freerouting are you using?

v1.8.0 (patched with https://github.com/freerouting/freerouting/pull/220 )

* [x] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?

Ubuntu 23.04, Linux 6.2.0-27, 64-bit

* [x] What EDA do you use? KiCad, EAGLE, other?

KiCad PCB Editor 6.0.11

* [ ] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?

DSN file, but I cannot attach it. I will work to create a minimal-reproducible PCB.

* [x] Did you use any command line arguments?

```bash
java -jar freerouting-executable.jar -l en -de pcb/board.dsn -do pcb/board.ses -mp 1000 -mt 20 -oit 0 -us hybrid -hr 1:1 -is random
```

* [x] Can you post a screenshot of the problem?

[image]

[image]

[image]

* [ ] Can you include the latest, related log entries from the .log file?

Not currently.

* [x] Please list the detailed steps to reproduce the problem

Used the command line as specified with the patch (although it would not make a difference in this case).

* [x] What would you expect freerouting to do in this situation?

Not place traces through holes or under existing exposed copper.

* [x] Is fixing this issue critical for your work?

It is a subtle issue that could be missed. I got a PCB manufactured with a dead short to ground.

The current workaround is to switch from `-is random` to `-is sequential`, so far this does not produce violations.

daniel-theia, 2023-08-30:
Possibly related tickets:

* https://github.com/freerouting/freerouting/issues/191
* https://github.com/freerouting/freerouting/issues/155

daniel-theia, 2023-08-31:
Trialling on a smaller version of the problem that runs faster (I will upload shortly), it could be that the violation is not kept as a viable solution, but is still displayed. It is unfortunate if this is the case, as for autorouting that requires more computation it can be a very long time before another potential solution is rendered - so it looks like it settles on a bad solution.

daniel-theia, 2023-08-31:
[image]

Obviously a problems in the above image: blue traces overlap behind the large red square.

```bash
java -jar freerouting-executable.jar -l en -de board.dsn -do board.ses -mp 1000 -mt $(np) -oit 0 -us hybrid -hr 1:1 -is random
```

In the terminal I see:

```
2023-08-31 10:47:33.678 [Thread-503] WARN  OptViaAlgo.opt_via_location: move via failed
2023-08-31 10:47:34.580 [Thread-554] WARN  OptViaAlgo.opt_via_location: move via failed
2023-08-31 10:47:51.161 [Thread-0] INFO  Saving '/tmp/freerouting-f874feab.frb'...
2023-08-31 10:47:57.338 [Thread-1863] WARN  OptViaAlgo.opt_via_location: move via failed
```

And it now appears to be 'stuck' on a bad solution.

daniel-theia, 2023-08-31:
Also to prove it is possible to pass DRC (for the traces):

[image]

daniel-theia, 2023-08-31:
[image]

It has come to a pretty similar solution, but still has this issue with crossing traces at the bottom and the warning message being spit out:

```bash
2023-08-31 11:34:42.896 [Thread-210361] WARN  OptViaAlgo.opt_via_location: move via failed
```

I can't manually drag it either.

daniel-theia, 2023-08-31:
[image]

Meanwhile, `-mt 1` solves this problem in 0.27ms:

```
2023-08-31 12:49:04.571 [main] INFO  Freerouting v1.8.0 (build-date: 2023-08-31)
2023-08-31 12:49:04.580 [main] INFO  Opening 'board.dsn'...
2023-08-31 12:49:05.099 [main] INFO  Opening 'board.rules'...
2023-08-31 12:49:05.100 [main] ERROR File 'board.rules' was not found.
2023-08-31 12:49:05.100 [main] ERROR File 'board.rules' was not found.
2023-08-31 12:49:07.545 [Thread-0] INFO  Starting auto-routing...
2023-08-31 12:49:07.683 [Thread-0] INFO  Saving '/tmp/freerouting-f874feab.frb'...
2023-08-31 12:49:07.818 [Thread-0] INFO  Auto-routing was completed in 0.27 seconds.
2023-08-31 12:49:07.825 [Thread-0] INFO  Saving 'board.ses'...
```

Definitely a big problem here in the threading.

andrasfuchs, 2023-10-03:
Thank you @daniel-theia for the detailed report!

Yes, multithreading is broken and it's likely to be permanently disabled in the future.

I'll add an extra warning message for now to discourage the usage of the `-mt` command line argument.

closed 2023-10-03 by andrasfuchs

daniel-theia, 2023-10-03:
> Thank you @daniel-theia for the detailed report!
> 
> Yes, multithreading is broken and it's likely to be permanently disabled in the future.
> 
> I'll add an extra warning message for now to discourage the usage of the `-mt` command line argument.

Any idea how it might be fixed? I have some complex boards that could do with any acceleration they can get. I did have a look through for a few hours, but never really got to the bottom of anything.

I'm still yet to fully understand what the approach in the routing software, in places it reads a lot like C code? If you have some high-level idea how it is searching this enormous space it might add some context when reading the code.

andrasfuchs, 2023-10-03:
I know, it's not easy at all, it challenges me as well. One of the problems is that I inherited this code a few years ago from the author who seems to be unreachable, so there are still blurry parts for me too.

Answering to your questions though: it doesn't use C code, it's Java only. 
It reads the data from the DSN files using the `SpecctraDsnFileReader` and we will get a object-representation of the board in the end. That representation consists of classes from the `designforms.specctra` folder.
The auto-router does its thing through the `AutorouteEngine` class and then we either run `BatchOptRoute` (single-threaded) or `BatchOptRouteMT` (multi-threaded) route optimizer.

The important takeaway here is that the auto-routing is always single-threaded, only the second stage, the route-optimization gets multi-threaded (if enabled). So your potential gain is limited.

I hope it helps somewhat!

C64HexCoder, 2023-10-03:
I keep recieving email even though i unsubscribed multiply times.

daniel-theia, 2023-10-03:
@C64HexCoder That's not something we control. You can check your GitHub settings again or set an email filter.

> I know, it's not easy at all, it challenges me as well. One of the problems is that I inherited this code a few years ago from the author who seems to be unreachable, so there are still blurry parts for me too.

Ah okay, you've done a great job so far it seems, so well done on that. It's a super valuable tool, it just needs some instability ironing out.

> Answering to your questions though: it doesn't use C code, it's Java only.

I understand that, I meant that it reads like it was original written in C, or that the original developer liked C.

> The important takeaway here is that the auto-routing is always single-threaded, only the second stage, the route-optimization gets multi-threaded (if enabled). So your potential gain is limited.

Interesting, thank you. Maybe I could take a look at `BatchOptRoute` and write a new multi-threading system around it. It would probably also teach me a lot about how this all actually works too.