freerouting/freerouting #222: I can't load the .DSN file

liu2-3zhi, 2023-08-31:
To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
V1.8.0
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
64—nit
* [ ] What EDA do you use? KiCad, EAGLE, other?
EasyEDA (https://easyeda.com/)
* [ ] What type of file are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
.DSN
* [ ] Did you use any command line arguments?
no parameters
* [ ] Can you post a screenshot of the problem?
look down
* [ ] Can you include the latest, related log entries from the .log file?
No
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
I hope it works fine
* [ ] Is fixing this issue critical for your work?
uncertain

WARNING Library.read_scope: app.freerouting.board padstack not found
[image]
[PCB13.zip](https://github.com/freerouting/freerouting/files/12487884/PCB13.zip)

liu2-3zhi, 2023-09-01:
DSN files can be loaded normally by konekt electra
[image]

EasyEDA2021, 2023-09-21:
please export your PCB design or project file to EasyEDA support, we will have a look
that should be EasyEDA issue, if you use EasyEDA Pro please try latest version, if you use LCEDA Pro, please update to v2.1.33

closed 2023-09-22 by andrasfuchs

liu2-3zhi, 2023-09-28:
I aleady use EasyEDA with the latest version