freerouting/freerouting #228: Automation: snapshot dialog prevents routing

andrasfuchs, 2023-09-27:
### Discussed in https://github.com/freerouting/freerouting/discussions/227

<div type='discussions-op-text'>

<sup>Originally posted by **saadxprt** September 26, 2023</sup>
i want to disable this feature 
[image]

im running free-routing with 
`java -jar freerouting-1.8.0.jar  -de ./MyBoard.dsn -do ./MyBoard.ses -dl `

how do i disable this feature? im running an automation, so unless i press on yes or no it will not work</div>

andrasfuchs, 2023-09-27:
You can't disable this feature at the moment, but you can delete the snapshot file(s) before running freerouting to prevent this dialog to show up.

I think you have a valid point, this dialog should have at least a timeout, defaulting to "no", or we could have a command line argument that makes freerouting ignore the snapshot files.

saadxprt, 2023-09-29:
can you tell me what I have do to delete snapshot file itself? 
Do I have to remove the prev .dsn & .ses file 
or
Do I have to remove log/freerouting.logs

andrasfuchs, 2023-10-03:
The snapshot files are typically in the temp folder of the system and they have the extension of ".frb". You can safely delete them and start the auto-routing again.

The curious thing is that this feature should be disabled by default, and it can only be enabled by using the `-im` command line argument. I'll double-check this, this is probably a bug.

andrasfuchs, 2023-10-03:
There was actually an issue with this feature, thank you for pointing it out!

I've just made a fix, it's going to be fixed in the next release.

closed 2023-10-03 by andrasfuchs

saadxprt, 2023-10-08:
thanks a lot waiting for it anxiously

andrasfuchs, 2023-10-10:
Could you please test the snapshot build?

First I would like to ask you to delete the `.frb` files from your temp directory and then try to run the latest SNAPSHOT build that is downloadable from here: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

Let me know if it still doesn't work as you would expect it so that I can fix it for the next release.

Thank you!

saadxprt, 2023-10-10:
I've tried it and it works exactly as it should, there is no more snapshot file prompt.

Could you please also do something like this for start-now prompt which automatically goes away after 20s
what if I want to disable 20s feature? can i include anything in the cli so it doesnt do that?
`java -jar freerouting-snapshot.jar -de ./MyBoard.dsn -do ./MyBoard.ses -dr ./MyBoard.rules`

andrasfuchs, 2023-10-10:
That's great to hear! 👍 

Let me check that command line argument thing for you. We don't have that option at the moment, but if it feels like that it can be done in an hour, then sure, no problem.

I'll get back to you here, if/when I implement it.

andrasfuchs, 2023-10-12:
We have now a new command line argument "-dct <seconds>" for setting the dialog confirmation timeout value. 
You can set it to 0 to disable the confirmation dialog.

saadxprt, 2023-10-12:
is this build also included in the new release? 
`freerouting-SNAPSHOT-20231012_142600.jar`

saadxprt, 2023-10-14:
can my routes go further into even more layers? except top and bottom layer when my board is complex