freerouting/freerouting #241: cli only in docker…?

hedefalk, 2023-10-27:
Version 1.8.0
docker/linux/64bit amd

I'm trying to understand how to use this in a pure cli env within a docker workflow. I'm trying to replace an old Dockerfile using freerouting 0.1.0 with 1.8.0.

Having a simple Dockerfile build:

```
FROM eclipse-temurin:17
RUN mkdir /opt/app
ADD https://github.com/freerouting/freerouting/releases/download/v1.8.0/freerouting-1.8.0.jar /opt/app
ENTRYPOINT [ "java", "-jar", "/opt/app/freerouting-1.8.0.jar"]
```

Then 

`docker run -it  --rm woodenstake/freerouting:v1.8.0 -help` works fine and prints the help but:

```
✗ docker run -v (pwd):/board -w /board --rm woodenstake/freerouting:v1.8.0 -de output/pcbs/board.dsn -do output/routed_pcbs/board.ses -mp 100
2023-10-27 20:51:31.147 [main] INFO  Freerouting v1.8.0 (build-date: 2023-05-22)
2023-10-27 20:51:31.315 [main] INFO  Opening 'output/pcbs/board.dsn'...

Exception: java.awt.HeadlessException thrown from the UncaughtExceptionHandler in thread "main"
```

Is there a mandatory set of params needed to the cli to _not_ open a ui? Do I need to supply a rules file?

hedefalk, 2023-10-28:
Trying:

`docker run -v (pwd):/board -w /board --rm --entrypoint java woodenstake/freerouting:v1.8.0 -Djava.awt.headless=false -jar /opt/app/freerouting-1.8.0.jar -de output/pcbs/board.dsn -do output/routed_pcbs/board.ses -mp 100 -dr freerouting.rules`

```
Exception in thread "main" java.lang.UnsatisfiedLinkError: /opt/java/openjdk/lib/libawt_xawt.so: libXext.so.6: cannot open shared object file: No such file or directory
	at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
	at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2394)
	at java.base/java.lang.Runtime.load0(Runtime.java:755)
	at java.base/java.lang.System.load(System.java:1953)
	at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
	at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
	at java.base/jdk.internal.loader.NativeLibraries.findFromPaths(NativeLibraries.java:315)
	at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:285)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2403)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:818)
	at java.base/java.lang.System.loadLibrary(System.java:1989)
	at java.desktop/java.awt.Toolkit$2.run(Toolkit.java:1388)
	at java.desktop/java.awt.Toolkit$2.run(Toolkit.java:1386)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
	at java.desktop/java.awt.Toolkit.loadLibraries(Toolkit.java:1385)
	at java.desktop/java.awt.Toolkit.initStatic(Toolkit.java:1423)
	at java.desktop/java.awt.Toolkit.<clinit>(Toolkit.java:1397)
	at java.desktop/java.awt.Component.<clinit>(Component.java:624)
```

hedefalk, 2023-10-28:
New approach courtesy https://stackoverflow.com/a/74303795/312873:

An image with x11vnc:

```
FROM ubuntu:latest
RUN apt-get update && apt-get install -y software-properties-common
RUN add-apt-repository universe
RUN apt-get update && apt-get upgrade -y
RUN apt-get install -y openjdk-17-jdk x11vnc xvfb
RUN mkdir ~/.vnc /app
RUN x11vnc -storepasswd 1234 ~/.vnc/passwd
COPY entrypoint.sh /entrypoint.sh
RUN mkdir /opt/app
ADD https://github.com/freerouting/freerouting/releases/download/v1.8.0/freerouting-1.8.0.jar /opt/app
RUN chmod +x entrypoint.sh
ENTRYPOINT ["/entrypoint.sh"]
```

Entrypoint:
```
#!/bin/bash
Xvfb :99 &
export DISPLAY=:99
x11vnc -forever -usepw -create &
java $@
```

Fails if I try to run java straight like:

    docker run -v (pwd):/board -w /board --rm --entrypoint java woodenstake/freeroutingx11:v1.8.0 -Djava.awt.headless=false -jar /opt/app/freerouting-1.8.0.jar -de output/pcbs/board.dsn -do output/routed_pcbs/board.ses -mp 100 -dr freerouting.rules

But I can run in with -it no problem, just need to wait for x11vnc to start:

```
  docker run -v (pwd):/board -w /board --rm -it --entrypoint bash woodenstake/freeroutingx11:v1.8.0
root@64272fe4562a:/board# Xvfb :99 &
[1] 9
root@64272fe4562a:/board# export DISPLAY=:99
root@64272fe4562a:/board# x11vnc -forever -usepw -create &
[2] 12
root@64272fe4562a:/board# 28/10/2023 08:22:32 -usepw: found /root/.vnc/passwd
28/10/2023 08:22:32 x11vnc version: 0.9.16 lastmod: 2019-01-05  pid: 12
28/10/2023 08:22:32 
28/10/2023 08:22:32 wait_for_client: WAIT:cmd=FINDCREATEDISPLAY-Xvfb
28/10/2023 08:22:32 
28/10/2023 08:22:32 initialize_screen: fb_depth/fb_bpp/fb_Bpl 24/32/2560
28/10/2023 08:22:32 
28/10/2023 08:22:32 Autoprobing TCP port 
28/10/2023 08:22:32 Autoprobing selected TCP port 5900
28/10/2023 08:22:32 Autoprobing TCP6 port 
28/10/2023 08:22:32 Autoprobing selected TCP6 port 5900
28/10/2023 08:22:32 listen6: bind: Address already in use
28/10/2023 08:22:32 Not listening on IPv6 interface.
28/10/2023 08:22:32 

The VNC desktop is:      64272fe4562a:0
PORT=5900

root@64272fe4562a:/board# 
root@64272fe4562a:/board# 
root@64272fe4562a:/board# java -Djava.awt.headless=false -jar /opt/app/freerouting-1.8.0.jar -de output/pcbs/board.dsn -do output/routed_pcbs/board.ses -mp 100 -dr freerouting.rules
2023-10-28 08:23:16.316 [main] INFO  Freerouting v1.8.0 (build-date: 2023-05-22)
2023-10-28 08:23:16.482 [main] INFO  Opening 'output/pcbs/board.dsn'...
2023-10-28 08:23:17.234 [main] INFO  Opening 'freerouting.rules'...
2023-10-28 08:23:17.260 [main] WARN  RulesFile.read: open bracket expected
2023-10-28 08:23:37.371 [Thread-0] INFO  Starting auto-routing...
2023-10-28 08:23:40.284 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:24:12.727 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:24:43.185 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:25:13.444 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:25:44.179 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:26:14.776 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:26:45.508 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:27:16.100 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:27:46.637 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:28:17.287 [Thread-0] INFO  Saving '/tmp/freerouting-82428e5.frb'...
2023-10-28 08:28:26.562 [Thread-0] INFO  Auto-routing was completed in 4 minute(s) 49.11 seconds.
2023-10-28 08:28:26.587 [Thread-0] INFO  Starting route optimization on 1 thread...
2023-10-28 08:28:33.275 [Thread-0] INFO  Route optimization was completed in 6.63 seconds.
2023-10-28 08:28:33.343 [Thread-0] INFO  Saving 'output/routed_pcbs/board.ses'...
```

I guess I'll just put a sleep command in that entrypoint for now but would be nice with something better

Or even better if there's a simple way to disable the ui alltogether…? :)

andrasfuchs, 2023-10-29:
Thanks @hedefalk for the detailed tests! 

> Or even better if there's a simple way to disable the ui alltogether…? :)

I agree 100%, we need to get to the point where we can run Freerouting without a GUI. The challenge is that the code is not prepared for it at the moment at all. Since it's one of the most requested features, I already started some preparations, and I plan to get to the headless version in the next 3-4 releases.

We are going to have a release next week, but it won't support that yet. The next one after that will be a "feature reduction" release, where I cut as much of the functionality as I can (based on usage statistics), and then I can start working towards the headless version, a docker container, and the RESTful web API interface.

I can't say anything about the timing though, since I work on this project in my free time, for free, so any help (suggestions, PRs, donations) are welcome (see #177 for details).

hedefalk, 2023-10-29:
@andrasfuchs Thank you for giving this old beast some love :)

Update: I could actually get by without any sleep - error was that I accidentally overrode the entrypoint above. So just:

```
#!/bin/bash
Xvfb :99 &
export DISPLAY=:99
x11vnc -forever -usepw -create &
# sleep 1 & # no need
java -jar /opt/app/freerouting-1.8.0.jar $@
```

works fine.

andrasfuchs, 2023-10-30:
Do I understand it correctly that you were able to run Freerouting in a Docker container without using its GUI?

If so, could you give me a step by step guide to replicate it? I'm sure it would save me a lot of time when I get to the point to support Docker!

ceoloide, 2023-11-08:
> Do I understand it correctly that you were able to run Freerouting in a Docker container without using its GUI?
> 
> If so, could you give me a step by step guide to replicate it? I'm sure it would save me a lot of time when I get to the point to support Docker!

I am doing the same in [my project](https://github.com/ceoloide/corney-island). Essentially I am using xvfb to run freerouting JAR executable:

```sh
xvfb-run -a java -jar freerouting-1.9.0.jar ...
```

closed 2024-01-15 by github-actions[bot]

hedefalk, 2024-01-24:
@andrasfuchs Sorry I never got back to you. Yes, the fake x11 approach works for me. I created a repo 

https://github.com/hedefalk/freerouting-x11

just now so I can publish that image and use it for my needs:
https://github.com/hedefalk/keydette/blob/ebd98154d7b13c36b5ba4aac0c2e0f3c17921652/Makefile#L25

hedefalk, 2024-01-24:
@ceoloide That looks neat! Should I possibly just replace my script here: https://github.com/hedefalk/freerouting-x11/blob/b19f4f7022cfe2a25f2559e9464f4c3c8d580d19/entrypoint.sh#L5

with just this oneliner?

hedefalk, 2024-01-25:
@ceoloide I couldn't get your version running, hm…

@andrasfuchs My approach seem to work fine for my use with freerouting v1.8.0, but not with v1.9.0. Not sure why, it just seems to hang…

andrasfuchs, 2024-01-25:
@hedefalk Thanks, that's a valuable feedback! 

I think the root of the issue can be two things: 
- we display a dialog (without an timeout) if the DSN file was exported from an older KiCad
- sometimes we get an error when we check for a new version and that is also a dialog on the GUI

I'll look into this when I get the time for the next release, sometime late February/March.