freerouting/freerouting #245: V.1.9.0 Unable to start

Bronc0, 2023-11-02:
Hey there,
I'm running a Win10x64 mashine,
Freerouting is running as plugin inside of Kicad V7.08.
When I try to startin freerouting v1.9.0 I can't get even the GUI,
When I change back to Freerouting v1.8.0 it works instantly.
I installed JRE 17,19&21.
Is there anywhere a logfile created, because I couldn't find.
Best regards

[image]
[image]

andrasfuchs, 2023-11-03:
Hi @Bronc0 , thanks for the report!

It looks like [it's a recursion error](https://www.reddit.com/r/learnpython/comments/enp7oy/what_does_3221225725_0xc00000fd_exit_code_means/), but I can't tell where it comes from without the stack trace.

Could you try to run `freerouing-1.9.0.jar` from the command line (eg. `java.exe -jar freerouting-1.9.0.jar -de MyBoard.dsn -do MyBoard.ses`)? It's located in `C:\Users\SeB\Documents\KiCad\7.0\3rdparty\plugins\app_freerouting_kicad-plugin\jar` as you can see on your screenshot. The log file will be generated automatically in the folder if it's not run as a plugin.

Bronc0, 2023-11-03:
Hi @andrasfuchs ,
I tried at first to start it from the Kicad GUI, error happened again (no log-file was created as you mentioned run as plugin).
I adjusted the command-line (freerouting worked as intended / Log-file)
[freerouting.log](https://github.com/freerouting/freerouting/files/13250818/freerouting.log)

java -jar freerouting-1.9.0.jar starts Freerouting without any issues

Is the plugin-button maybe misslinked?

andrasfuchs, 2023-11-03:
Thank you for the confirmation, I wasn't sure if we have an issue with the Java or the Python part of the execution.

It looks like we made a mistake in the Python file (which is the worse scenario regarding the debugging process). 
Fortunately though we have just a few places where we can cause a recursion error/infinite loop, but I will still need your help to figure this out, because I don't have this error, and I don't know how to reproduce it.

I took a quick look into the source code, and I can't find the most likely place for this to happen. I would suspect that it's something to do with multiple version of Java installed, or something else with the directories in your temp folder.

Anyway, I really need to focus on other projects today and in the following weeks, so I'm sorry for the inconvenience, I'll get back to you when I can spare a few hours for this.

Bronc0, 2023-11-03:
Thank you for your quick action,
When I installed V1.9, I had only one JRE, but due to that error-message I tried some other.
When you have time for a guess, I can try for sure

closed 2024-02-12 by github-actions[bot]