freerouting/freerouting #247: java.lang.NoClassDefFoundError when open freerouting

EasyEDA2021, 2023-11-13:
### Please provide as many details as you can
* What version of freerouting are you using?
v1.9.0
* On what platform did you run it? Windows11,  64-bit

[image]

when I start the client, it appear this error, I try uninstall and re-install, it still same, anyone can help how to solve this issue?

thanks

ProfessorKa0Z, 2023-11-14:
I get the same error with 1.9.0 on MacOS 13.5.2 (M1):
[image]
It doesn't seem to affect function though.

Nikolai521, 2023-11-16:
Same on win7 64 bits
[image]

SpiderKiCad, 2023-11-22:
Same issue. freerouting 1.9.0, Windows 10 x64, jre-8u391-windows-x64 installed:

[image]

tomthecarrot, 2023-12-27:
Same issue on macOS 13.6.2 with OpenJDK 17.

andrasfuchs, 2024-01-25:
@Lycosaurus That's a great piece of information, I'll take a look when I get to work on the next release. Thanks!

jsiddall, 2024-03-19:
Also seeing this on V1.9.0 on Linux (since no one else seemed to be running that in the posts I read)

codegsaini, 2024-04-16:
v1.9.0 same issue on linux 

[image]

parasyte, 2024-08-15:
It's still a problem. While I hate to add nothing of value with my comment, I don't want to see issues like these closed just because no one is working on them.

jsiddall, 2024-08-15:
According to issue #288 it is resolved in 2.0

parasyte, 2024-08-15:
That's good for a future where 2.0 is available.

andrasfuchs, 2024-08-15:
[Version 2.0 is in the works](https://github.com/freerouting/freerouting/issues/326) and will be released later. You can experiment with [the snapshot build](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT) until then.

fabianlefebre, 2024-09-17:
Thanks for the fix!
I am not seeing the issue in the 2.0 snapshot build but I run into a  NoSuchElementException instead when starting the auto router. Could of course be unrelated. (MacOS, java version 21.0.4)

andrasfuchs, 2024-09-19:
@fabianlefebre It's probably unrelated, but still important. Could you please create a new issue with the details, so that I can fix it before the release? I haven't seen that exception during my tests yet.

fabianlefebre, 2024-09-19:
@andrasfuchs done! #348

closed 2024-11-06 by andrasfuchs