freerouting/freerouting #252: Error while reading DipTrace file

galacticstudios, 2023-11-21:
When loading the attached DSN file, Freerouting gives the error:
```
WARNING DsnFile.read_string_scope: closing bracket expected at ''
ERROR   There was an error while reading DSN file.

```

Freerouting v1.9.0 (build-date 2023-10-30)
Running on Windows 10, 64 bit
DSN file was created by DipTrace
[Wang.zip](https://github.com/freerouting/freerouting/files/13430793/Wang.zip)

galacticstudios, 2023-11-21:
I found the problem, on line 127. There was a space in the symbol name.

Another routing program spotted the error and reported the line number. So I guess this bug is now about providing better error messages, e.g. with a line and column number.

galacticstudios, 2023-11-21:
I take that back (sorry). After fixing the problem at line 127, the error still occurs in Freerouting. 

The other routing program (DeepPCB) accepts the file, but gives warnings that some components lie partially outside of the board. (That's intentional.)
[Wang.zip](https://github.com/freerouting/freerouting/files/13430887/Wang.zip)

Here's the new file that shows the problem.

ProfessorKa0Z, 2023-11-23:
I noted this problem in an earlier bug report.  You should be able to work around it by changing the single-quotes (') in lines 4 and 5 of your file to double-quotes (").