freerouting/freerouting #261: Add design rule files to import easyeda JSON files

mayjack0312, 2024-01-15:
### Please provide as many details as you can
* What version of freerouting are you using?
I am using version 1.9.0

* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
I am using Windows 64-bit

* What EDA do you use? KiCad, KiCad with freerouting plugin, EAGLE, EasyEDA, other? What is its exact version?
I am using EasyEDA

* What types of files are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
[Config_DesignRule_PCB1_2024-01-15.json](https://github.com/freerouting/freerouting/files/13937139/Config_DesignRule_PCB1_2024-01-15.json)

* Did you use any command line arguments?
I have not modified any parameters

* Please list the detailed steps to reproduce the problem
When using the software normally, the design rules are not the default design rules of easyeda

* What would you expect freerouting to do in this situation?
Added JSON file function that can directly import easyeda design rules
For example, I want to import the above Config_DesignRule_PCB1_2024-01-15.json

andrasfuchs, 2024-01-25:
I like the idea, I just can't tell how much work it would be. 
I'll get back to this with the next release, if I can spare the time (probably in March).

mayjack0312, 2024-01-25:
> I like the idea, I just can't tell how much work it would be. I'll get back to this with the next release, if I can spare the time (probably in March).

Okay, hopefully this will be a useful feature. Let us patiently wait for the arrival of new features

andrasfuchs, 2024-03-04:
I contacted EasyEDA and asked for the JSON specs today, I hope they can help me out.

I'll keep you guys posted here.

andrasfuchs, 2024-03-05:
@Lycosaurus Interesting insight, thank you! I will test this myself soon.

Summving (@EasyEDA2021) from EasyEDA already replied to my inquiry, and provided me this page for the JSON file clarification: [EasyEDA Pro Tutorial / Design - Design Rule](https://prodocs.easyeda.com/en/pcb/design-design-rule/index.html)

closed 2024-05-11 by github-actions[bot]

Gifford47, 2025-04-01:
> [EasyEDA Pro V2.1.49 64-bit win10 Freerouting V1.9.0[
> 
> I have Freerouting now working with the design rules from EasyEDA. When you first open the design with Freerouting, you need to say NO to importing the design rules. What Freerouting is really asking is to import the local FR design rules. When you say NO, you get the EEDA design rules imported. This is assuming you first opened FR with the .jar file. (not FR.exe).
> 
> There is still a bug it seems with EEDA (or maybe with FR). When you choose the JLCPCB design rules, when the design is imported back to EEDA (I use metric) from FR then the DRC will find some spacing errors (I did not change it to metric) which are very small. I think there is a conversion issue between metric and imperial dimensions. I set the DRC with the stricter design rules, then when I import back from FR I do a DRC with the JLCPCB design rules. No DRC errors.
> 
> I'm curious to see if this works for others as well.
> 
> P.S. There is a pull-down on the bottom left of the EEDA design rules where you can easly select from different pre-configured design rules, as well as creating your own design rules file.

hi @Lycosaurus !
how do you import the rules in the new version v2.0.1? I always get many clearance violations...

Lycosaurus, 2025-04-01:
> [EasyEDA Pro V2.1.49 64-bit win10 Freerouting V1.9.0[
> I have Freerouting now working with the design rules from EasyEDA. When you first open the design with Freerouting, you need to say NO to [...]

> 
> hi [@Lycosaurus](https://github.com/Lycosaurus) ! how do you import the rules in the new version v2.0.1? I always get many clearance violations...

Hi Gifford47,

I have not used CAD or Freerouting for about a year, so cannot give you an update on the newer version.  What I found to be a workaround at the time is what I described in the first paragraph above. I take it that by saying NO (to importing design rules) in Freerouting, you are not using the local Freerouting rules, but rather, the rules you developed in your CAD layout are used (which is what you want). Conversely, a YES will use the local Freerouting rules. I hope that helps.