freerouting/freerouting #264: routes at the bottom layer miss  when import .ses file

lbsonggz, 2024-02-07:
I  designed the PCB with LCeda and then output the .dsn file. And I autoroute the PCB with the freerouting-1.9.0 jar in Java  
environment and the PCB is autorouted successfully. Then I export the .ses file. But when I import the .ses file in LCeda，only the routes at the top layer is appeared. I want to know the reason.
 
[image]
[image]

andrasfuchs, 2024-03-04:
@lbsonggz Could you attach your DSN file or KiCad project files?

Is your bottom layer is defined as "power" layer by any chance? There is a similar issue here: #269

lbsonggz, 2024-03-15:
I have tried  freerouting for many times, then I find it is mainly because there are too many routers including vias and the freerouting software only saves some routes on the top layer, especially when it runs with more than 2 threads and you stop it manually. It is neither the bug of the freerouting software nor the LCEDA software. If it reports finishing routing, I can save the right .ses file, but some vias are missing. So can I give you some suggestions?  
1)Adding one parameter, saving interval, that the freerouting software can save the completed routers at this interval? e.g., 2min.
2)Adding the priority control？ that is, it permits some routers,e.g. UART are routed at higher priority comparing to other route.

closed 2024-05-21 by github-actions[bot]