freerouting/freerouting #269: Cannot route auto route bottom layer

lucasdietrich, 2024-02-19:
I'm experiencing an issue where freerouting doesn't use the bottom layer of my board. When running the program it seems to only route the top layer without using any via, even if components are present on both sides. As my board is not trivial, it fails to route the board. Moreover, all options related to the bottom layer are disabled.

I'm new to Kicad and freerouting, there is a chance I misconfigured my Kicad project.
Thank you for your help

* What version of freerouting are you using?
**Freerouting version 1.19**

* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
**Windows 11**

* What EDA do you use? KiCad, KiCad with freerouting plugin, EAGLE, EasyEDA, other? What is its exact version?
**Kicad 7.0.10**

* What types of files are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
[freerouting.dsn.zip](https://github.com/freerouting/freerouting/files/14336387/freerouting.dsn.zip)

* Did you use any command line arguments?
**No**

* Can you post a screenshot of the problem?
[image]
[image]

* Can you include the latest, related log entries from the .log file?
**Can't find this file.**

* Please list the detailed steps to reproduce the problem
Created the project under Kicad

* What would you expect freerouting to do in this situation?
Top and bottom layers are routed

* Is fixing this issue critical for your work?
Yes

iveyj1, 2024-02-28:
I am having the same issue, identical symptoms - no vias, no backplane use, incomplete routing.  Same os, version 1.19.  A PCB that routed properly in 1.18 doesn't work in 1.19.  No improvement exporting dsn and running freerouting from the command line.  Dsn available on request.  Thank you.

andrasfuchs, 2024-03-04:
I quickly look into your DSN file and I think the reason is that your B.Cu layer is defined as "power" type. 
Freerouting allows routing only on "signal" types.

Would you expect another behavior? If yes, could you help me find the definition for the _signal_ and _power_ layer types and explain the difference between the two? Should the routing algorithm act differently on these layers?

iveyj1, 2024-03-04:
I was mistaken about my symptoms, there is not a problem with the bottom layer, it is with the power planes.  I confirmed that the middle two layers of the board are marked for power.

Using Kicad 8 integration.

I hacked in version 1.7.0 by copying to the plugins directory.  It routes as expected - only a few missing traces:

[image]

With the identical project, using version 1.9.0 most or all plane connections are missed:

[image]

I couldn't find the .ses and .dsn used by the integration, but I attached a separately-exported .dsn file (this is the one that doesn't work).

[example dsn.zip](https://github.com/freerouting/freerouting/files/14487205/example.dsn.zip)

iveyj1, 2024-03-04:
To answer your question, I don't expect traces on a power plane, but there should be via connections.  See for example U302 (the topmost 4-pin component on the board)  The center two pins are ground and +3.3V. 
Here is the way it should be, with +3.3 and ground connected through vias to the center planes.
[image]
And here with those connections left as airwires.
[image]

By the way, the only routing parameter I changed was to reduce the via cost from 50 to 5.

iveyj1, 2024-03-07:
> CAUTION: This email originated from outside the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> I quickly looked into your DSN file and I think the reason is that your B.Cu layer is defined as "power" type.

I can open a separate issue if that helps.

Thanks,
Iveyj1

@andrasfuchs<https://github.com/andrasfuchs> You are correct, I mistakenly misconfigured by bottom layer as a power plane. After changing it to "signal", I was able to auto route both sides as expected.

This addresses my initial question, however @iveyj1<https://github.com/iveyj1> seems to have pointed out another issue. Should I keep the issue open ?

andrasfuchs, 2024-03-07:
@lucasdietrich It's no wonder you didn't realize that that's the reason why the layers are disabled, our UI isn't obvious at all. Sorry about that, I have just added a tooltip for the disabled layers, explaining that power layers are disabled for routing.

@iveyj1 Thank you, but no need, I will rename this issue now and will investigate it in the next few weeks.

andrasfuchs, 2024-04-16:
@iveyj1 I tried to open the file you provided, and I got an error regarding ascii characters. I dig deeper, and it looks like KiCad 8 exported the DSN with an error at line 1844. 

`
(class kicad_default "" +5V +Vin_prot "/0-10_interface/0-10V_PWM_OUT_#1"
`

The double-quotes at the beginning of the net class definition is incorrectly would mean that there is a net with the name of "". There isn't in the DSN and reading the list of nets fail.

When I remove the "" from before the +5V, it loads correctly, and I think Freerouting can also wire the board as it should.

Could you test it yourself as well to make sure?

iveyj1, 2024-04-18:
It seems to load, route, and complete, however the connections to the planes are still not made.

By the way, it appears I inadvertently attached a DSN file  that had already been correctly routed (presumably by an earlier version of freerouting).  That one already has the correct connections to the plane.  Sorry for the confusion.

I'm not 100% sure I'm invoking freerouting correctly.  What I did to test was:
In Kicad 8 PCB, delete traces and vias
Export DSN file
Delete the "" in line 1844 of the DSN
In the plugin directory run:
C:\Users\ivey\Documents\KiCad\8.0\3rdparty\plugins\app_freerouting_kicad-plugin\jar>java -jar freerouting-1.9.0.jar
Then in freerouting, open the modified DSN file and start routing (I'm not sure how to properly invoke to open the file directly).  
[image]
The +3.3 and ground traces are not connected.

iveyj1, 2024-04-18:
I made a much smaller example from scratch using default settings and get the same results.  The extra quotes are present, however removing them does not cause correct routing.

This is the result from 1.8.0 and is routed correctly:
[image]

This is the result from 1.9.0 with the quotes removed - ground not routed even though ground plane is available.
[image]

I attached the relevant files, thanks for your help with this.
[min_fr_test.zip](https://github.com/freerouting/freerouting/files/15030866/min_fr_test.zip)

pi1541, 2024-05-29:
I also have this problem, only when updating to freerouting-1.9.0-windows-x64.msi
It will never drop a via into an internal power plane.

andrasfuchs, 2024-06-05:
@iveyj1 The .zip you attached doesn't have any files.

andrasfuchs, 2024-06-05:
I have just made a tiny board to test the issue, but I couldn't reproduce the problem.

KiCad 8 board setup:
[image]

In Freerouting 2.0 the power layers are disabled:
[image]

Autorouter runs without any issues.

When I import the .ses file into KiCad again, the via looks good to me:
[image]

What do I miss?

iveyj1, 2024-06-05:
Sorry, here's the file with the simplified example.  
[min_fr_test.zip](https://github.com/user-attachments/files/15594460/min_fr_test.zip)

andrasfuchs, 2024-06-07:
@iveyj1 Thank you, I got closer to the root of the issue! 

I could finally reproduce the problem.

As I investigated further, I suspect that there is something wrong with the net association of the plane when we read the DSN file.

This looks correct:
[image]

But the conduction area has this proeperty:
[image]

I think it's off, and it should be a one-item array that has a value of 0 (representing GND) instead of the 1 we have.

I'm not sure yet if that's it, I just wanted to give a quick update.

iveyj1, 2024-10-08:
Hey @andrasfuchs , did you ever find out what the trouble is?

closed 2025-03-15 by github-actions[bot]