freerouting/freerouting #283: Board has many small unconnected traces under pads

jsiddall, 2024-03-19:
### Please provide as many details as you can
* What version of freerouting are you using?

v1.9.0 2023-10-30

* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?

Linux 64 bit

* What EDA do you use? KiCad, KiCad with freerouting plugin, EAGLE, EasyEDA, other? What is its exact version?

KiCad 6.0.10

* What types of files are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?

Exported DSN file, can attach if it would help

* Did you use any command line arguments?

No

* Can you post a screenshot of the problem?

Yes:

[image]

* Can you include the latest, related log entries from the .log file?

Not sure it is relevant, but there were several messages on the console including:

2024-03-19 11:52:03.519 [Thread-153] WARN  FloatLine: Parameter is null
2024-03-19 11:52:12.137 [Thread-153] WARN  CalcFromSide: start corner was not found

* Please list the detailed steps to reproduce the problem

Generate DSN in KiCad, autoroute, export SES, re-import into KiCad

* What would you expect freerouting to do in this situation?

Generate the layout as it did but don't leave tracks with unconnected ends (flagged as DRC warnings)

* Is fixing this issue critical for your work?

No, freerouting is great even with this inconvenience.  Keep up the great work!

andrasfuchs, 2024-03-19:
Thanks @jsiddall for the report!

Could you attach the KiCad project files so that I can reproduce the whole workflow?

jsiddall, 2024-03-19:
Update: having looked at these suprious tracks, they all seem to be on power and ground pads.  This may be relevant since there were a small amount of power and ground traces done in KiCad before sending the DSN to freerouting

jsiddall, 2024-03-19:
> Thanks @jsiddall for the report!
> 
> Could you attach the KiCad project files so that I can reproduce the whole workflow?

I would have to cleanup the project first.  It has the traces added by freerouting but the folder also has a lot of other older versions in it.  I can do that if it would help, though the DSN would be a lot easier.

Also, could I email the files to you directly?  Would rather not post publicly.

andrasfuchs, 2024-03-19:
We could start with the DSN if that's easier for you, I might be able to find the bug that way.

Please send it to info@freerouting.app. Would it be alright with you if I uploaded the DSN into this repo for testing purposes just like [the others](https://github.com/freerouting/freerouting/tree/master/tests)?

Thank you!

jsiddall, 2024-03-19:
I emailed the cleaned up KiCad project, DSN and SES files.  Let me know if you need anything else.

Thanks!

jsiddall, 2024-05-20:
I see the bot is getting impatient!  Any progress on this?

closed 2024-09-25 by github-actions[bot]