freerouting/freerouting #289: We may need a optimize of freerouting!

L1uTongweiNewAccount, 2024-03-26:
### Please provide as many details as you can
* What version of freerouting are you using?
v1.9.0 (build date: 2023-10-30)
* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Windows 64-bit
* What EDA do you use? KiCad, KiCad with freerouting plugin, EAGLE, EasyEDA, other? What is its exact version?
EasyEDA
* What types of files are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file?
the exported DSN file
* Did you use any command line arguments?
no, just the shortcut by freerouting
* Can you post a screenshot of the problem?
[image]
[image]
* Can you include the latest, related log entries from the .log file?
no
* Please list the detailed steps to reproduce the problem
just start routing
* What would you expect freerouting to do in this situation?
optimize the code of routing.
Look the pictures above here. It just use the 11% of CPU, and it's not enough.
We need use multi-process to optimize it. If we can rewite it by C++ and make an assembler optimize, it's better.
After all, I need 100% of CPU used to speed my process, because it's really slow with 1000+ network.
* Is fixing this issue critical for your work?
yes, a bit.

L1uTongweiNewAccount, 2024-03-26:
[image]
[image]

andrasfuchs, 2024-04-03:
Hi @L1uTongweiNewAccount , thank you for the report!

Could you provide your big DSN file? I would need to investigate the issue further, because there is a chance that the first stage (auto-router) works, but the second stage (router optimizer) gets stuck.

You can also send it to me to info@freerouting.app, if you don't want to attach it here.

L1uTongweiNewAccount, 2024-04-04:
@andrasfuchs Fine, here is it!
[Autorouter_PCB_FHT-VGA_2024-03-25.txt](https://github.com/freerouting/freerouting/files/14865847/Autorouter_PCB_FHT-VGA_2024-03-25.txt)

[Autorouter_PCB_FHT-8086_2024-03-08.txt](https://github.com/freerouting/freerouting/files/14865845/Autorouter_PCB_FHT-8086_2024-03-08.txt)

Of course, you should change it to .dsn because i can't upload .dsn files. And they are meaningless after i found design errors in them... But they are good examples for test!

Thanks!

L1uTongweiNewAccount, 2024-04-04:
And if i think routing optimizer is slow, i can stop it. but i cannot export .ses with no routing error because it has not been done.

I think we can have a option to skip optimizer.

andrasfuchs, 2024-04-05:
Excellent, thank you @L1uTongweiNewAccount!

Optimizer can be disabled, but the setting isn't easy to find (I will improve the UI/UX soon): 
[image]

[image]

Up in the menu, Settings, Auto-router, remove the check from the "Optimization".

L1uTongweiNewAccount, 2024-04-05:
Thanks a lot!

andrasfuchs, 2024-04-12:
I can confirm that running the auto-router with your VGA file needed me 2.5GB of RAM and 4 hours 46 minutes. 
Both seem unreasonably high, so this needs some investigation.

Thank you again for the files, they help a lot with the testing!

I don't think I will be able to fix this just yet, but I'll look into this after I get [the new scoring system](https://github.com/freerouting/freerouting/issues/287) in place.

EasyEDA2021, 2024-04-30:
> Excellent, thank you @L1uTongweiNewAccount!
>
> Optimizer can be disabled, but the setting isn't easy to find (I will improve the UI/UX soon): [...]

Hi, where can I download the v2.0 snapshot? thanks

andrasfuchs, 2024-04-30:
The current snapshot version is available here: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

L1uTongweiNewAccount, 2024-05-24:
I am thinking that if we can divide the board into some parts and alloc them to some defferent threads, and link them in the end.

I'm not sure how to divide them, and how to link them together. But I think a big PCB can be Recursively divided.

Will this idea helps to you? @andrasfuchs

GorillaSapiens, 2024-09-24:
> I am thinking that if we can divide the board into some parts and alloc them to some defferent threads, and link them in the end.
> 
> I'm not sure how to divide them, and how to link them together. But I think a big PCB can be Recursively divided.
> 
> Will this idea helps to you? @andrasfuchs

Here are some common strategies used in multithreaded PCB routing algorithms:

1. Net Partitioning
The routing problem can be split by partitioning the nets, where each thread works on routing a subset of the nets. This is particularly useful when there are many nets to route, as it reduces contention and allows each thread to focus on its subset. Ensuring that threads don't interfere with each other by routing over the same space is key to success here.

Challenges: Conflict resolution between threads that may attempt to route nets over the same areas of the PCB.

2. Layer-Based Partitioning
Another approach is to assign different threads to work on different layers of the PCB. This reduces the chances of interference between threads while providing a natural partition of the problem. Each thread can focus on optimizing the connections on its assigned layer.

Challenges: Ensuring that vias (connections between layers) are placed efficiently without conflicts between the threads.

3. Adaptive Maze Solvers (Dijkstra or A-based)*
Advanced algorithms like A* or Dijkstra’s can be parallelized to route multiple nets simultaneously. In these algorithms, each thread can solve the shortest path for a different net, avoiding interference by ensuring that a net that is actively being routed locks the resources (e.g., tracks and vias) it occupies.

Challenges: Synchronization overhead when locking routing resources to prevent collisions or bottlenecks.

4. Simulated Annealing (SA) and Genetic Algorithms (GA)
These optimization algorithms are used for global routing, where multiple solutions are explored simultaneously. In multithreaded implementations:

Simulated Annealing: Each thread explores a different potential routing solution by "cooling" the system at different rates. The threads periodically share their best solutions.

Genetic Algorithms: Threads evaluate and evolve different populations of routing solutions. They periodically exchange information to converge on an optimal routing layout.

Challenges: Ensuring convergence to a good solution without too much synchronization overhead.

5. Concurrent Rip-up and Reroute
Rip-up and reroute is a method where previously routed nets are selectively removed and re-routed to avoid conflicts. In a multithreaded implementation, threads can work on re-routing different parts of the PCB concurrently, reducing congestion hotspots.

Challenges: Coordination is required to ensure that threads do not rip up and reroute the same regions at the same time, which could lead to conflicting solutions.

6. Decomposition of Routing Grid
The routing grid of the PCB can be divided into smaller sub-regions, each of which can be routed independently by different threads. After each thread completes its task, the regions are stitched together.

Challenges: Ensuring smooth handoff between sub-regions to avoid routing inconsistencies at the boundaries.

7. OpenMP and Task Parallelism
Some algorithms take advantage of OpenMP (or similar libraries) to distribute tasks dynamically. The routing process is divided into independent tasks, and threads execute those tasks in parallel. This can work well when different parts of the routing problem (e.g., evaluating routing cost, updating routing paths) can be executed concurrently.

Challenges: Task dependencies and load balancing can become issues if some threads finish earlier than others.

L1uTongweiNewAccount, 2024-09-24:
@GorillaSapiens That's really what I'm thinking! Is it possible if you write a faster algorithm to solve this problem?

Now I'm trying to make it multi-threading and support more EDA software, but I don't understand Java in fact.

If someone write a tool, just input a DSN and output a SES with processing, it will give me a lot help.

GorillaSapiens, 2024-09-26:
> If someone write a tool in C++, just input a DSN and output a SES with processing, it will give me a lot help.

Java is not all that different from C++.  if someone wrote the tool you are describing in C++, 99% of the work would be done, and porting it to Java would be trivial.

GorillaSapiens, 2024-10-01:
a possible braindead solution might be to just use the same algorithm on multiple cores.  the algorithm could use lots of random numbers (e.g. deciding the order in which to route nets, or which nets to rip up, etc...).  as each core comes up with a solution, the result is scored and placed into a pool.  cores which have completed a solution can (again at random) choose top scoring solutions from the pool, and attempt to optimize them (again more randomness).  of course each core would start with a different RNG seed.  not saying this is a GREAT solution, but it wouldn't be hard to implement.

L1uTongweiNewAccount, 2024-10-01:
@GorillaSapiens are you talking about Simulated Annealing? Good idea.

In fact, many algorithms can be multicored. Breadth First Search has a queue, so that it can provide jobs to different cores.

I don't know what algorithm did freerouting use. If it can be multicored, that will be good.