freerouting/freerouting #298: Error: unable to access jarfile freerouting-1.9.0.jar

summerfind, 2024-04-18:
### Please provide as many details as you can
* What version of freerouting are you using?
*  freerouting-1.9.0.jar

* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Windows
* What EDA do you use? KiCad, KiCad with freerouting plugin, EAGLE, EasyEDA, other? What is its exact version?
 EasyEDA
* What types of files are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file? (Please note that GitHub doesn't allow DSN file uploads, so you need to ZIP it before uploading.)

[image]

not support .zip ? and freeRouting not on yet!
* Did you use any command line arguments?
no
* Can you post a screenshot of the problem?
[image]

* Can you include the latest, related log entries from the .log file?

* Please list the detailed steps to reproduce the problem
* installed Freerouting, installed java jre, click Freerouting

* What would you expect freerouting to do in this situation?
be able to run Freerouting? 
* Is fixing this issue critical for your work?
no

andrasfuchs, 2024-04-18:
@summerfind Could you replicate the problem with the latest snapshot release? I think it is fixed, but it would be great to have confirmation from you.

You can download the snapshot jar file from here: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

summerfind, 2024-04-18:
> @summerfind Could you replicate the problem with the latest snapshot release? I think it is fixed, but it would be great to have confirmation from you.
> 
> You can download the snapshot jar file from here: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

Thanks, 
I used  freerouting-1.9.0.jar to open Freerouting OK now.
and I try to modified the issue as unfinished routing, and I can't upload .zip and even picture.
there is small board maybe squeezed too much? the Auto-router pass: #183 now. 
it works when all wire set 0.254mm, if set VCC 1mm or 0.75mm never done.

L1uTongweiNewAccount, 2024-05-18:
Very like my issue #299, also a never finish problem

I-Dont-Want-To-Create-An-Account, 2024-08-20:
I don't know if it helps, but I'm also seeing this problem

closed 2025-03-15 by github-actions[bot]