freerouting/freerouting #299: Strange & silly issue when routing

L1uTongweiNewAccount, 2024-04-23:
[Autorouter_PCB_MCRDO-5_2024-04-23.txt](https://github.com/freerouting/freerouting/files/15080965/Autorouter_PCB_MCRDO-5_2024-04-23.txt)

I attached my radio project with .txt(but you need rename it to .dsn) for you.

When width=0.5mm & spacing=0.25mm, i found it falls into a strange loop with the network Q1_3.

But in fact, it just can routing by human easily without any holes. So I think maybe it can reflect some algorithm bug or disadvantages.

You can see the first time, this system is quickly stopped with "auto-routing completed" but we can found network Q1_3 was bad.

And we try to restart the auto-routing, but we falls into a infinty loop with increasing processing number.

but if we just look for Q1_3, we will find it just need a simple connect without any magic.

L1uTongweiNewAccount, 2024-04-23:
Maybe a special crashed judge program thinks the circult without Q1_3 is also correct ... ?

In my computer, sometimes it will loop infinty, sometimes it will just enter optimizer and stop the auto-routing without routing Q1_3.

So I just thinking: Did freerouting not see Q1_3 was incorrect? Or it think there's no solution of it?

Anyway, there must be a bug. And I think it was a impartant bug.

Djzhan0207, 2024-04-24:
I will figure it out. Maybe before 6 month，plz wait

L1uTongweiNewAccount, 2024-04-24:
@Djzhan0207 Good. And also see #289, some test DSN files may be useful to you.

谢谢老哥了！

closed 2024-09-27 by github-actions[bot]