freerouting/freerouting #300: The LCEDA/EasyEDA exported DSN file cannot read the copper pour

wcx-cau, 2024-05-07:
### Please provide as many details as you can
* What version of freerouting are you using?
V1.9.0
* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Windows 11 64-bit
* What EDA do you use? KiCad, KiCad with freerouting plugin, EAGLE, EasyEDA, other? What is its exact version?
 LCEDA/EasyEDA
* What types of files are you working with? Can you attach the project files (.pro, .kicad_pcb, etc) or the exported DSN file? (Please note that GitHub doesn't allow DSN file uploads, so you need to ZIP it before uploading.)
exported DSN file
* Did you use any command line arguments?
no
* Can you post a screenshot of the problem?

* Can you include the latest, related log entries from the .log file?

* Please list the detailed steps to reproduce the problem

* What would you expect freerouting to do in this situation?
可以识别铺铜区域，避免重复连接
* Is fixing this issue critical for your work?

L1uTongweiNewAccount, 2024-06-12:
I met the same issue, too.

L1uTongweiNewAccount, 2024-06-12:
See issue #152.

closed 2024-10-17 by github-actions[bot]