freerouting/freerouting #305: Inner layers route through vias

mrkylegp, 2024-05-18:
Easyeda
Windows 11
Freerouting v1.9.0

Freerouting routes tracks for inner layers through vias, causing violations.

[image]
[image]

Gifford47, 2024-07-22:
Do you already have a solution for that? I've exactly the same problem for all of my PCBs.

Gifford47, 2024-07-24:
@andrasfuchs do you any idea how to fix that error or get a walkaround?

andrasfuchs, 2024-07-25:
Hey guys, sorry for not reacting sooner.

@Gifford47 I don't think I will have time to focus on algorithmic issues like this before the v2.0 release, there are [plenty of other things I would like to complete](https://github.com/freerouting/freerouting/issues/326) in the next few weeks.

Having said that I think it's an important issue, so just to make sure I understand it correctly: if we have let's say a 4 layer board and we need to connect tracks on layer 1 and layer 2 using a (blind) via, Freerouting creates a via that goes trough all layers, not just layer 1 and 2, and that's the main issue?

I would urge you (or anyone else from the community) to try to fix this, I would be glad to review a pull request and merge it into the codebase before the v2.0 release, I just don't have the time now to dig deep into this. If it doesn't happen, it's very likely will be only part of one of the releases after that, meaning you have to wait for this for months. 

If you need assistance with getting started with the fix, let me know!

mrkylegp, 2024-07-26:
> Hey guys, sorry for not reacting sooner.
>
> @Gifford47 I don't think I will have time to focus on algorithmic issues like this before the v2.0 release, there are [plenty of other things I would like [...]

Thanks for the response! These are just through hole vias (all the way through the board), not blind/buried vias. The inner layers route straight through them as seen in the original post image (green inner layer).

Gifford47, 2024-07-26:
> If you need assistance with getting started with the fix, let me know!

I would like to try to find the error first, but I only have basic knowledge of java. Can you possibly point to the files or functions that could be responsible for the error? Then I would try to work my way into it...

andrasfuchs, 2024-07-26:
We have the [Via](https://github.com/freerouting/freerouting/blob/master/src/main/java/app/freerouting/board/Via.java) and [DrillItem](https://github.com/freerouting/freerouting/blob/master/src/main/java/app/freerouting/board/DrillItem.java) classes in the app.freerouting.board package, I would start there and look for any methods that add a new instance of them.

Please publish your findings here or in a pull request, and thank you for your efforts!

Gifford47, 2024-07-26:
ok, the first thing i found is, that through hole vias are not right allocated to the corresponding layers:
[image]
- the padstack p11e9 is a hole and was correctly allocated
- the padstack p12e14 is a through hole via and was only allocated to layer 1 and 2, which is incorrect. is must be also placed on layer 15 (inner layer 2) and 16 (bottom)

therefore the other tracks on the lower layers (15,16) are routed "through" the via because there is no via anymore :)

andrasfuchs, 2024-07-26:
@Gifford47 Nice catch! Does this problem occur with both circular and rectangular padstacks, or is it always with the rectangular one?

Gifford47, 2024-07-26:
It seems that it occurs on the circular and the rectangular padstacks. The strange thing is, that the vias in my pictures above are coming directly from the PCB editor(easyeda). They aren't generated by freerouting app.

So the vias above are wrong processed right after the import. Instead of throughhole vias they are only blind vias anymore.
Is this possible? Does nobody had this issue with multilayer PCBs before ?

I will take a step deeper the next days...

Gifford47, 2024-07-30:
> It seems that it occurs on the circular and the rectangular padstacks. The strange thing is, that the vias in my pictures above are coming directly from the PCB editor(easyeda). They aren't generated by freerouting app.
> 
> So the vias above are wrong processed right after the import. Instead of throughhole vias they are only blind vias anymore.
> Is this possible? Does nobody had this issue with multilayer PCBs before ?
> 
> I will take a step deeper the next days...
> 
> Btw: the via option with dropdown menu would be a cool feature!

Hey! Short update: 

This is not an issue on freerouting side. It's an issue in easyeda side by using the .dsn file export extension. It's generally designed for 2 layer PCBs so it does have a bug when exporting 4 layer PCBs with multiple layers. I've contacted easyeda support to fix it. The following is the response of the easyeda support:
"After verification, we found out that the problem does exist.
It is a bug issue, which we have documented and will fix it later.
Thank you for informing us."

Therefore the issue can be closed. @andrasfuchs thanks for the support and this great project!!

andrasfuchs, 2024-07-30:
I'm glad it's sorted, thank you for your investigation @Gifford47!

closed 2024-07-30 by andrasfuchs

Gifford47, 2024-07-31:
i actually wrote a (quick and dirty) script which fixes an easyeda 4layer pcb with TH vias saved as .dsn file.
run the python (>=3.9) script in the same folder as the *.dsn file and voilà, it converted 2 layer vias into 4 layer vias. 1 layer pads will be untouched.

```
# coding=utf8

import re
import glob

regex = r"\(shape\([^)]*\)"
top_layer = '1'
bottom_layer = '2'
multi_layer_no = ('15', '16')

#######################################################################

file_list = glob.glob('./*.dsn')

for input_file in file_list:
    output_file = input_file
    f = open(input_file, "r")
    lines = f.readlines()

    for count, line in enumerate(lines):
        matches = re.finditer(regex, line, re.MULTILINE)
        #print("Line{}: {}".format(count, line.strip()))

        for matchNum, match in enumerate(matches, start=1):
            print("Match {matchNum} was found at {start}-{end}: {match}".format(matchNum=matchNum, start=match.start(),
                                                                                end=match.end(), match=match.group()))
            layer = line.split(' ')
            r = [i for i in layer if i in multi_layer_no]
            if r:
                # line is already in all layers (through hole)
                print('This line is already multilayer-ready! Skipping ...')
                continue
            else:
                # if next line contains only the end of the section... (only ')')
                if re.match(r"\s+\)", lines[count+1]):
                    # line has to be copied to additional layers defined in 'multi_layer_no'
                    try:
                        index = layer.index(bottom_layer)       # get element in list with last layer (bottom)
                        for i in multi_layer_no:
                            new_layer = layer.copy()        # copy last line/layer
                            new_layer[index] = i            # change layer number to multilayer number
                            lines.insert(count+1, ' '.join(new_layer))  # insert new layer line into original list of lines
                            count += 1
                    except Exception as e:
                        print('Probably only SMD pad, skipping ...')
                        continue
            # for groupNum in range(0, len(match.groups())):
            #     groupNum = groupNum + 1
            #     print("Group {groupNum} found at {start}-{end}: {group}".format(groupNum=groupNum, start=match.start(groupNum),
            #                                                                     end=match.end(groupNum),
            #                                                                     group=match.group(groupNum)))

    with open(output_file, 'w') as out:
        for line in lines:
            out.write(f"{line}")

```