freerouting/freerouting #311: Command Line Interface / HTTP Interface needed

L1uTongweiNewAccount, 2024-06-12:
I know devlopers always see freerouting as a design platfrom, but in face it is often not very easy to use and most people just export .dsn  in their own EDA to route,

So I think if we can provide CLI and HTTP(s) Interface for EDA and Linux users (who doesn't have a desktop environment) .

- HTTP Interface can refer [EasyEDA Std Tutorial](https://docs.lceda.cn/en/PCB/Route/index.html#Local%20Auto%20Router) .

In this example, EasyEDA's auto-router has own HTTP Interface to call by EasyEDA (EasyEDA based on electron, more like a browser).

In most case EasyEDA's auto-router doesn't do well like freerouting, so maybe analyse the network activity from EasyEDA can provide a native routing support for it. :)

- CLI Interface can improve from now, just add --no-gui option and --interactive or --no-Interactive for human or program.

It just like in interactive:

```
$ java -jar freerouting.jar --no-gui --interactive
Welcome to freerouting v1.8.0.
Freerouting is a free software, xxxxxxx
Type 'help' if you need help.
> help
......
> config optimize false
> auto-routing
Auto-routing started.
Auto-routing \#1 was running, xxx routed, xxx to be route, xxx failed.
Auto-routing \#1 was running, xxx routed, xxx to be route, xxx failed.\
......
```

or in no-interactive:

```
$ java -jar freerouting.jar --no-gui --no-interactive --auto-routing --no-optimize
Welcome to freerouting v1.8.0.
Freerouting is a free software, xxxxxxx
Auto-routing started.
Auto-routing \#1 was running, xxx routed, xxx to be route, xxx failed.
Auto-routing \#1 was running, xxx routed, xxx to be route, xxx failed.
```

Maybe CLI will be cooler and faster than now GUI, I think.

seveibar, 2024-06-17:
Just wanted to weigh in- I've been looking at trying to wrap freerouting in an http server, but I was thinking about just wrapping the CLI for simplicity, since it's quite a bit of work and a bit of a different model to have a daemon directly inside of freerouting IMO. I've been looking online a bit but it doesn't seem like there's been an effort yet to create a freerouting. Can follow up here if/when I make some progress!

L1uTongweiNewAccount, 2024-06-17:
> Just wanted to weigh in- I've been looking at trying to wrap freerouting in an http server, but I was thinking about just wrapping the CLI for simplicity, 

If you have warpped it in CLI, we can easily rewrap this CLI in a HTTP server.

Anyway, to compatible with easyda's HTTP Server, I am finding its API and I will open-source it if easyeda allows.

L1uTongweiNewAccount, 2024-06-18:
I got EasyEDA's router API in https://github.com/L1uTongweiNewAccount/EasyEDA-router-API just then.

I think this is a good example.

seveibar, 2024-06-18:
@L1uTongweiNewAccount nice find! Compatibility with the EasyEDA API is a good idea, although it does seem like websockets might be a bit difficult to use, we might want to expose an alternative API that is just HTTP based. I think we can match compatibility with JLCPCB though, thanks for the documentation!

L1uTongweiNewAccount, 2024-06-18:
@seveibar in fact just like what I say, many CAD supports plugins. HTTP interface can let us integrate to them to make us more convenient.

andrasfuchs, 2024-06-18:
@L1uTongweiNewAccount The idea is great, I had similar plans as I described here: https://github.com/freerouting/freerouting/discussions/193

I have made a few changes to version 2.0 that I think will make it possible to hack together a web API version until we get a proper implementation. I haven't tested it yet, but @hedefalk could make Freerouting work in headless mode with an earlier version: https://github.com/freerouting/freerouting/issues/241

Our current snapshot version potentially works again, it worth a try: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

L1uTongweiNewAccount, 2024-06-21:
@andrasfuchs snapshot has some acceptable small problems...

Anyway, I think #241 is discussing about run freerouting in fake x11 desktop and we are not able to run a headless version yet.

I came up with a crazy idea just like #241, emulate a vnc desktop and emulate mouse and keyboard in GUI, but it is obviously complex and foolish :)

To sum up, if we can write a headless, as least the following characteristics should be possessed:

- it doesn't require any desktop including emulate VNC desktop, etc.
- it can give more clearly feedback more than `[Thread-0] INFO  Saving xxx...`, and it would better has progress.
- it can run as interactive mode.

I will hack a headless mod, and i will call it as `freeroutingd` (Freerouting daemon).

L1uTongweiNewAccount, 2024-06-21:
I'm trying to use Xvfb: https://www.scivision.dev/xvfb-on-windows-subsystem-for-linux/

This is a fake X11 desktop.

L1uTongweiNewAccount, 2024-06-21:
@andrasfuchs I accept the first test, my script works well.

`freeroutingd` is a bash script, and it will create a fake X11 window to run freerouting.

Next, I will optimize the command line (because i think the origin cli options are a bit strange)

andrasfuchs, 2024-06-23:
@L1uTongweiNewAccount 
This is great, I'll try to take a closer look at your PR next week. 👍 

The [EasyEDA Std Tutorial: Local-Auto-Router](https://docs.lceda.cn/en/PCB/Route/index.html#Local-Auto-Router) link that you provided above was not reachable from Europe, but here is the alternative URL that works: https://docs.easyeda.com/en/PCB/Route/index.html#Local-Auto-Router

Do you know about a detailed documentation about the API that we could mimic in Freerouting to make the interface simple to integrate to EasyEDA? I would prefer something like [a Swagger page](https://swagger.io/tools/swagger-ui/), but any documentation would help. Thanks!

L1uTongweiNewAccount, 2024-06-23:
@andrasfuchs  ok, I will update the documention.

L1uTongweiNewAccount, 2024-06-29:
I got a latest solution with #315 & #316.

L1uTongweiNewAccount, 2024-06-30:
@andrasfuchs I already published my package `freerouting-launcher` in npm. (https://github.com/L1uTongweiNewAccount/freerouting-launcher)

seveibar, 2024-06-30:
@L1uTongweiNewAccount wow that's pretty cool!!!!! ❤️

andrasfuchs, 2024-07-03:
Cool stuff! Thanks for your contribution! 🥇 

I added `freerouting-launcher` as a submodul in the [integrations/EasyEDA folder](https://github.com/freerouting/freerouting/tree/master/integrations/EasyEDA), so others can find it 👍

L1uTongweiNewAccount, 2024-07-03:
@andrasfuchs For sure, the log format I descibed in #315 must be done, so that the additonal features in freerouting-launcher can be enabled.

For now, use `--old` option in launcher does well.

andrasfuchs, 2024-07-03:
I have just modified the logging a little bit, we have logs like this (with the default INFO log level):

[image]

Does this look good?

If you change the log level with `-ll TRACE` you can see more details about the process, but the number of incompletes are not available at this point in the log.

L1uTongweiNewAccount, 2024-07-03:
@andrasfuchs In fact I mentioned this in DEBUG level.

Now, Everything what have progress all based on "incompletes", this is EasyEDA's standrand, and it is important to display percents.

If we can guess a limit of passes and optimizer passes, maybe these are enough good.

Anyway, we need a method to guess the percent, although it can be inaccurate.