freerouting/freerouting #319: Freerouting won't start fom kicad 8.0.3

AuBeTeX, 2024-07-08:
Greeting!

I had a problem with freerouting. Java cannot be found when launched from Kicad. There are plenty of them installed. By default, the oracle version is above. When the error occurred, I installed the recommended adoptium.net version, then openjdk, which can be installed by Microsoft powershell. Java is available when run from the command line. It's included in Path. I downloaded all java, I only installed version 17 from adoptium, but it still didn't work. Although the description says 17 or newer.

The freerouting jar itself works when launched from the console. but I don't want to use it from there. I looked at the plugin.py file and although I don't know much about python, it looks like it's only looking for version 17.

Is it possible to somehow put it in the plugin.ini where the java is? If so, how? I don't know because there are too few comments in the plugin.ini. I don't know how to enter the format, provided that the system uses it.

I think it may be caused by some kind of update or installation, since it was working with Oracle's Java about 1 month ago. But since a lot has happened since then, I can't go back. I am looking for an alternative to repair. And it would be nice to know what caused it. After all, it can happen to other people (and it did, according to the forums. There are people who have solved the problem by reinstalling Java. Not for me.)

data:
Freerouting version: 1.9.0
Platform Windows 10 Pro 22H2 19045.4529
EDA: Kicad with freerouting plugin, installed from plugin manager. Ver.: 8.0.3
I don't use command line arguments, I wouldn't use them from the command line, I don't know how to set them in kidcad.
Image:
[image]
There is no log file because the autorouter does not start

Reproduce: Open Kicad -> Open PCBnew -> click autorouter icon.

I'm expect: Run and work :) 

* Is fixing this issue critical for your work?  : It would significantly speed up development.

andrasfuchs, 2024-07-10:
I would try the following:

1. Let the KiCad plugin try to download and install a Java 17 JRE and run Freerouting. You can do this by clicking on "Igen" in the dialog you had screenshot of. (You probably already tried this, and it somehow failed.)

2. You can set the path of the java executable manually in the 11st line of plugin.ini, eg:
`path = 'c:\\Program Files\\Eclipse Adoptium\\jdk-21.0.2.13-hotspot\\bin\\java.exe'`

3. If that fails too, you can still try to uninstall KiCad and reinstall it without using their preferred Flatpak method. Flatpak isolates the plugins that makes JRE inaccessible for Freerouting.

4. If all those fail, check if the _default_ Java installation is the Adoptum one

5.  As a last resort I would try to temporarily disable any antivirus software if you have any installed, they can interfere with Java executables

AuBeTeX, 2024-07-11:
Greeting!

Kicad cannot install java throws an error message, see below.
[image]

I entered the line in the ini as you suggested, it was not interested.

I don't understand the installation situation, Kicad is normally installed in administrator mode (for all users) with the windows installer, freerouting was installed with the package manager and it worked. I have since uninstalled freerouting and reinstalled it with the package manager. Uninstalling Kicad would cause a problem, I applied a lot of unique parts to it. I will be able to reinstall it when I finish this circuit.

I uninstalled all Java and only adoptium is up, first 17 and then 21. It doesn't work with either.

As long as it was working, there was an antivirus, but only the Windows one. But I tried it and it didn't help.

I'm going to try reinstalling the kicad, but now I have to wire this cable by hand.

bum3bg, 2024-07-17:
I have had KiCad version 7 and freerouting stopped working (same error message)

andrasfuchs, 2024-07-19:
@bum3bg Do I understand that Freerouting worked before for you? 
If so, do you have any guesses what happened on your system just before it stopped working?

bum3bg, 2024-07-19:
Yes, it worked. Not sure what happened. Maybe Kicad extension was updated in the meantime. Note that I can still run standalone jar version of Freerouting. To me it seems that check for the JRE is broken. It complains about Flatpak although it is Kicad on Windows.

AuBeTeX, 2024-07-19:
Hello!

It worked for me too, although I regularly updated java and Kicad as well. I don't know what happened to me either, but in principle neither windows nor Java nor Kicad was updated between the working and non-working ones. Two or 3 days passed. That's it. Now, however, I do not like to reinstall the Kicad because I have put a lot of unique parts into the Kicad. My biggest regret is that I can't somehow synchronize the Kicado on several computers at the level of components and settings, because the autorouter still works on my laptop, for example.

andrasfuchs, 2024-07-19:
I would suspect a JRE location issue as well, despite the fact that you must have a valid JRE installation since you can run the standalone app. KiCad might have made the environment where plugins run even more restrictive? It needs more testing.

The Flatpack message on Windows is misleading, I agree, but it shouldn't interfere with the JRE search function.

andrasfuchs, 2024-07-19:
@AuBeTeX Thanks for the extra info, that helps finding the reason a little. Are the KiCad and Java versions exactly the same on both machines you have? Are the JRE installation located in the exact same folder? Could you check the temp folder on both computers and check (1) if you have a freerouting.json file there, (2) if you have any `jdk-*` folder in there?

If you have any of those, could you try to remove them one by one, and check if that fixes the issue?

You don't need to reinstall KiCad, I don't think that would help.

AuBeTeX, 2024-07-19:
Hello!

I deleted the contents of my temp folder, there were 2 jre folders in it, both of them were 17, only the subversion was different. Freerouting threw the first error, then the freerouter starts when checked, but when exited it throws the kicad error. I didn't say that, although it might have been important, but I store my files on my own computer, but I synchronize them with the drive. It throws the below error.
[image]

but the freerouter started. So, that problem was solved.

andrasfuchs, 2024-07-19:
Köszi @AuBeTeX!

It seems like the original error was fixed by deleting the JRE folders, that's good news. 👍 

The screenshots you pasted show a different issue, my suspicion is that we don't handle paths with non-ascii characters well. Could you move your input file from the "Saját meghajtó" folder to somewhere else, where there are no `á` and `ó` characters in the filename and path?

bum3bg, 2024-07-23:
@andrasfuchs I've deleted only freerouting.json from temp and plugin now starts.

andrasfuchs, 2024-07-23:
@bum3bg Thank you, that information is a great addon to this issue.

Dongsup-Seo, 2024-09-10:
Thank you. 
The Java version error is resolved by deleting the JRE folders from the %temp% directory.